mxtommy/Kip #258: Feature request. average wind direction.

petsicola, 2023-12-20:
add average wind direction gauge something similar as the opencpn gauge

[image]

B&G has something similar with average wind direction and average wind speed.
[image]

add gauge like this app with distance to start line. Sailgrib RS is the app. There are different apps but this is very simple.

[image]

godind, 2023-12-20:
Awesome!

godind, 2024-01-08:
Slack user request detail: The graphical instrument shows the average wind and its deviations to both sides.

This instrument is also used to control and display the result of a background processing average wind calculation using Tactics double exponential smoothing algorithm.

It is recommended to have it in your instrument catalog if you plan to use any other instrument requiring the information about the average wind. i give you a picture in the message
https://dashboard-tactics-pi.readthedocs.io/en/latest/tactics/tactics.html

godind, 2025-08-15:
and Wind Trend as of 3.8

closed 2025-08-15 by godind