mxtommy/Kip #262: The numbers is not visble on the compass on the widget autopilot

rezder, 2023-12-21:
When using the theme moderne dark the background and text color of is too similar.  I believe it happened when the autopilot got its own text color on #202

godind, 2023-12-22:
Send a screenshot. It looks fine on my side

godind, 2023-12-22:
Make sure you are on the latest version please. This is the only version supported. If you are trying to get up to date with your fork, you should rebase. You are 200+ commits behind. Save yourself some work ;)

rezder, 2023-12-22:
my master is from 19/12 and my rebase is from 26/11 this is a screnshot
[image]
the numbers is missing. You need two text classes to handle two different background light-ring numbers (120) dark-ring text (NW) you have only one. And I am testing my rebase now.

godind, 2023-12-22:
This dark color is because the button are disabled. It's intentionnal.

Thank for reporting!

closed 2023-12-22 by godind

rezder, 2023-12-22:
come on where the arrow is pointing the number 30 is missing thank you fro listenting

rezder, 2023-12-22:
The numbers are on the  light ring same color as text NW but you can not see them  30 60 90 120 150 where are they??. There are even space for them. Tell me the color class of the html text elements that holds the numbers 30 60  and the color.

rezder, 2023-12-22:
[image]
This is the template look there is number on it