mxtommy/Kip #268: Missing numbers on compass ring on auto pilot

rezder, 2023-12-22:
Theme with numbers on the ring:
[image]
Theme without
[image]
se earlier for more info

rezder, 2023-12-22:
The problem is the use of apText in svg-autopilot.component.html
on text elements with numbers.

<text class="apText"
            id="use7035"
            y="-448.97495"
            x="188.02534"
            style="font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;font-size:19.04160881px;line-height:1.25;font-family:sans-serif;font-variant-ligatures:normal;font-variant-caps:normal;font-variant-numeric:normal;font-feature-settings:normal;text-align:start;letter-spacing:0px;word-spacing:0px;writing-mode:lr-tb;text-anchor:start;display:inline;stroke:none;stroke-width:1.02008593"
            xml:space="preserve"
            transform="matrix(0,1.0002794,-0.99972065,0,0,0)"><tspan
              id="tspan7105"
              x="188.02534"
              y="-448.97495"
              style="stroke-width:1.02008593">90</tspan></text>

The numbers is on the ring where there are dashes
With hard coded black

<path
            id="use6664"
            d="m 98.386053,135.48218 14.374497,5.23484"
            style="display:inline;fill:#000000;fill-opacity:1;stroke:#000000

Black is the better choise for that ring but class will be better

The ring is <ellipse class="compass-inner-ring"

godind, 2023-12-22:
Thanks for the feedback! I'll fix it.

There are plans to align the both Wind and AP dials to the same styles and also build the AP Widget with a responsive design which is lacking at the moment. One contributor said he was working on something but I've not seen his design sketches yet.

Also note that the Material styles need to be fixed to upgrade to angular 16+. It's a bit messy at the moment. We also plan to remove some themes to reduce code base and styling complexity in widgets. Most themes are not used.

godind, 2024-01-13:
now part of v2.6

closed 2024-01-13 by godind