mxtommy/Kip #269: Move all deps to devDepencies

tkurki, 2023-12-26:
I don't think Kip needs any of the dependencies install when it is installed by end users?

So all of these https://github.com/mxtommy/Kip/blob/master/package.json#L74-L100 should go under devDeps. This would save a bit of install time and people's disk space.

godind, 2023-12-28:
How do you determine what goes under dev or not? I was thinking only dependencies that are used to code such as @types, testing and such.

tkurki, 2023-12-28:
Kip is a webapp that runs in the browser. Usually webapps are packaged to files that are under the plugin's `public` directory - all the dependencies are used at build or test time, none at runtime. 

So you can safely make **everything** devDependency.

If a plugin has a `public` nothing outside of it is available to the browser, which means that nothing that gets installed under `node_modules` from dependencies is ever even available to Kip running in the browser.

This is what I see in server access log when loading Kip, nothing from the dependencies:
```
GET /@mxtommy/kip/ 304 1.237 ms - -
GET /@mxtommy/kip/assets/tween-min.js 304 1.028 ms - -
GET /@mxtommy/kip/assets/steelseries-min.js 304 1.460 ms - -
GET /@mxtommy/kip/assets/NoSleep.min.js 304 1.916 ms - -
GET /@mxtommy/kip/runtime.e75056c32c37f59d.js 304 0.896 ms - -
GET /@mxtommy/kip/polyfills.64db6be9cdc8f808.js 304 0.606 ms - -
GET /@mxtommy/kip/assets/fontawesome6/css/all.min.css 304 1.106 ms - -
GET /@mxtommy/kip/styles.a5d5fc1506885765.css 304 0.611 ms - -
GET /@mxtommy/kip/main.026c51a2aa2a2ac8.js 200 4.448 ms - -
GET /signalk/ 304 0.356 ms - -
GET /@mxtommy/kip/assets/notification.mp3 304 0.493 ms - -
GET /@mxtommy/kip/assets/fontawesome6/webfonts/fa-solid-900.woff2 200 2.067 ms - 126828
GET /@mxtommy/kip/manifest.json 304 0.387 ms - -
GET /@mxtommy/kip/assets/manifest-icon-192.maskable.png 304 0.560 ms - -
GET /@mxtommy/kip/assets/screenshot-720x540.png 200 0.742 ms - 229447
```

PR #290 (2024-01-13): Move all deps to devDepencies
Fixes #269
commit b3996157 (2024-01-13): Move all deps to devDepencies
Fixes #269
closed 2024-01-13 by godind