mxtommy/Kip #270: Support for SK history api

tkurki, 2023-12-26:
Both signalk-to-influxdb and [signalk-to-influxdb2](https://github.com/tkurki/signalk-to-influxdb2?tab=readme-ov-file#retrieve-logged-data) support Signal K History API. I started adding the openapi description for the [history API](https://github.com/SignalK/signalk-server/pull/1653).

By using the history API Kip could retrieve the data for the historical datasets and start with full graphs instead of starting to accumulate the data when Kip is loaded.

History API would also allow navigating the graph datasets, back and forth in time and zooming in and out.

godind, 2024-01-17:
How does that work technically?
Storage is in InfluxDb, integrated is handled by an SK plugin and it's all exposed as the SK Playback API ?

tkurki, 2024-01-17:
Otherwise correct but not the Playback api, that is different. See the link in my previous post.

godind, 2024-01-17:
More for personal knowledge: is history a replacement for playback? I'd love to understand the different purposes/roadmap for each.

tkurki, 2024-01-18:
The difference should be pretty clear from the description?

Playback = ”hey server start playing back all the data from time xx onwards”
History api = ”hey server give me data for paths x and y from time xx to yy”

godind, 2024-01-18:
That's it? Playback sounds simpler and better suited for KIP.

tkurki, 2024-01-18:
Playback is for streaming all data in real time or slowed or speeded up, not for retrieving specific time series data.

irjudson, 2024-03-30:
The subtle difference is in data boundary parameters *and* return style (stream vs batch).
This might/could simplify *all* widgets by creating a new "simple" widget "graph over time" that connects to the history api and just displays "now - graph size" queries every update.

that's different from "playback my route from last monday" -- at least to me, but, does that make it any more or less clear to you @godind ?

PS - I'm about to try and join the active signalk/kip dev - is there any discord like community for real-time discussions>

godind, 2024-03-30:
Hi! Join the signal K #KIP Discord channel. We can chat there. I'll take all the help I can!

godind, 2024-03-30:
@irjudson lets chat on Discord. Im not sure I understand.

tkurki, 2025-04-27:
@godind now that Kip 3 is out maybe have a look at this?

So the basic idea is that when Kip loads it tries to fill in the datasets from the history API so that the graphs don't start empty but history is populated right away.

tkurki, 2025-09-09:
#782

tkurki, 2026-01-28:
Good points!

The endpoints return 404 if there are is no active history api provider plugin. And we need to also cater for older sk server versions that predate History API. See https://demo.signalk.org/signalk/v2/api/history/values

I don't think this should depend on specific plugins. For all I know there could be more plugins that implement History API and maintaining a list of them and their version ranges that have Histoary API support within KIP makes no sense to me. A practical example would be that somebody adds History API support to signalk-to-influxdb (for Influxdb v1).

> should not have to add, remove, tune every single path manually

It seems that you don't have experience with the history  plugins? Maybe you should give them a go. Infludb2 plugin's default setting is to store everything. Parquet  plugin works differently, default setting is to store nothing.

To me there are several "Service Levels":

---

1. Works transparently

If History API & data for the path is available everything works, no configuration needed in KIP.

2. Helps user to activate History API

Provides instructions on how to enable History API and activate storing history.

3. History API configuration integration

When a chart widget is configured uses History API 1.1 (to be implemented) to make sure that history is accumulated.

---
I fear 3 will bring in quite a bit of complexity. How data gets stored, how long it is kept, is older history decimated differently than recent. And if adding a widget activates storing data for the path does removing the widget stop it? 

One "escape hatch" I can think of would be that we add to the API a way for KIP to retrieve the path to the History API provider plugin's configuration screen. Then Kip could present a link to it and delegate plugin specific configuration to the plugin's configuration UI. So level 2, enhanced.

Imho I'd be happy with 1, 2 is easy, 2+ very doable but 3 takes us to "complex, lots of work" territory. I'd love to postpone that.

godind, 2026-01-29:
> The endpoints return 404 if there are is no active history api provider plugin...I don't think this should depend on specific plugins.

Agreed. It's a better approach.

> It seems that you don't have experience with the history  plugins?

I have zero experience with it and was counting on the interested parties to team up with me and help design and build. It's more efficient than me becoming expert in each plugin.

> Maybe you should give them a go. Infludb2 plugin's default setting is to store everything. Parquet  plugin works differently, default setting is to store nothing.

This will create UX challenges.

I'll reply more. On a snowmobile trip at the moment.

motamman, 2026-02-02:
Currently, signalk-parquet's data is debounced to 1 second. The History API implementation supports automatic time bucketing when none is specified, as well as manual buckets. It never returns all the data. (Though it could, I suppose, with a minor tweak.) 

Can return EMA, SMA, Median, Mean, Min, Max, First and Last based on the bucket size.

tkurki, 2026-02-02:
I don’t think we should get hung up on how exactly the current implementations work. I have talkes with Maurice about some future enhancements to parquet plugin.

What we should concentrate on is the user’s needs - what is the intent in making a history api call? Then it is up to the db plugin implementor how to best serve the request. Tomorrow somebody adds history api support to their choice of db and they may choose their strategy for persistence and aggregation.

godind, 2026-02-03:
> I don’t think we should get hung up on how exactly the current implementations work. I have talkes with Maurice about some future enhancements to parquet plugin.

@tkurki what is the minimal resolution for Influx?

tkurki, 2026-02-14:
ema and sma are now included in [OpenApi spec](https://demo.signalk.io/doc/openapi/?urls.primaryName=history#/default/get_values) and plugin api of server v2.22.0. signalk-to-influxdb2@2.0.0 includes support for them.