mxtommy/Kip #271: [object Object] ierror message

tkurki, 2023-12-26:
Navigating to settings shows me this

[image]

and does not really tell the end user anything useful. So maybe worth just removing that error message?

godind, 2023-12-27:
How did you get to this error? Just a plain install and going to settings?

tkurki, 2023-12-27:
Yes.

godind, 2024-01-12:
I tried to reproduce with no luck:
1. from clean install and simply going to: Configuration / Setting / Storage
2. from clean install, adding the connection to the server, and simply going to: Configuration / Setting / Storage 
3. from clean install, adding the connection with user authentication to the server, and simply going to: Configuration / Setting / Storage

Can you share setup detail and reproduction steps? I see from the Error message you shared that it's related to remote storage / configuration storage.

- What SK permission level does you user have?
- Do you have any files in: applicationData/users/<user>/kip ?
- Do you have any files in: applicationData/global//kip ? 

Thanks

tkurki, 2024-01-12:
- nothing in applicationData
- cleared browser localStorage
- load Kip
- configure server url
- unlock layout, add one numeric display, lock layout
- select Configuration - Settings

godind, 2024-01-12:
You are not authenticating to SK. Just connection with a URL, right?

tkurki, 2024-01-12:
Security on the server is off.

godind, 2024-01-14:
You mean all options are off? Like so?
[image]

Disabling `Allow Readonly Access`  does give me the same error. It's caused by an Application Data Storage Interface call that fails since Readonly is off and you are not authenticated.

KIP requires either `Allow Readonly Access` or authenticating with a user, and `Application Data Storage` Interface to be enabled. Is there a way to check these setting using some remote API call?

tkurki, 2024-01-15:
No, I mean server's *security* mechanism is *OFF*: it has not been enabled. There is no login in the admin ui top right and trying to access security settings gives you the `Enable security`screen.

PR #294 (2024-01-16): [object Object] error message
Fixes #271 with an improved error message and possible resolution guidance.
commit cf0d00cd (2024-01-16): [object Object] error message
Fixes #271
closed 2024-01-16 by godind