mxtommy/Kip #275: It is not possible to save wind widget with paths errors

rezder, 2023-12-29:
It should be possible to use the wind widget without all paths valid. Personally I was planning to use the waypoint and I have no wind vane. You can not save the widget configuration without all path is valid.
It was introduced on v2.4.2 package update.

Before there was another problem. If you saved a configuration with a invalid path the configuration data got corrupted as all disable fields like convertUnitTo and source got removed from the configuration. When the configuration was load again the missing fields coursed problems and prevented any path from being saved in a spot with missing fields.

The errors got court by the debug line:
https://github.com/mxtommy/Kip/blob/3867b29c17845c3b9bc7dfbe8b5a432d1ed682e2/src/app/modal-path-selector/modal-path-selector.component.ts#L95
and the error happen because it is not possible to enable a undefined field.

Its also caused other errors in the console. I believe in in the line:
https://github.com/mxtommy/Kip/blob/3867b29c17845c3b9bc7dfbe8b5a432d1ed682e2/src/app/signalk.service.ts#L530
but I am the sure of the line but there were other errors do to missing sourse on configurations.

Waypoint paths are a general issue for the validation as they are not always present. It also poses a problem for the timeout configuration as it is per widget not path. You do not want to configure a time out on a waypoint but you may want to on other paths.

godind, 2023-12-30:
Hi

At the moment, wind the widget is essentially designed so sailing and so wind instruments are required.

True that the waypoint can disappear if no waypoint is active. I had not thought of that. I will simply change it to Not configurable. That will hide it from the paths Options tab, making the form valid either way. It should fix the problem.

I agree that having timeout at the root of the widget is not ideal, but only the Wind/sailing widget uses more then one path at the moment. So it's not really a problem. I could also remove that option from the wind widget options, because as you mention, it will drop the waypoint value, and other values, if enabled. I'm working on a multi path Boolean widget at the moment so I might get to it but it's a priority. Add a separate feature request for this in the issues please.

Thank you

rezder, 2023-12-30:
Fair enough but are you not forgetting autopilot it have a lot of paths or is it not in active use. I think the restriction on widget to only have active paths is very limiting for big widgets which can be both god or bad. Thank you for listening.

godind, 2023-12-30:
Oh. True about the AP! I know of a few users whom are using it. Hard to say how many but I've helped some getting the setup. They report everything works and requested a few enhancements.

I'm not saying TTL and sampleTime should not be improved. Just that I did the basics for what I needed at that moment. Each contributor adds to KIP based on what they need or sends fixes that annoys them. We (MxTommy and I) help with the setup, architecture, code review and PR merge.

Just keeping up to date with Angular versions is hard work. KIP is on 14 and we need to get it to 17 which has a lot of new good features such as Standalone Components that would simplify widget creation, but the code needs to be upgraded first.

The roadmap for now is to first add two new PUT widgets, fix known v2 issues, improve the connection handling and add KIP as a SK default installed app.

Secondly is to upgrade angular and work on the framework. To get there we need to Fix the Sass (.scss) styles setup used in KIP.

It's all messed up the way it's used. It works but it generates a 1mb css output file! I think the way it's used KIP is loading the same styles over and over with a mix of @include and @mixing everywhere. I'm really not good at with Sass processing...

This would really simplify KIP, make it faster and allow to upgrade to angular 15 and beyond.

Want to help with Sass?

rezder, 2024-01-02:
Hi David

I do not know if I should reply to all on these mails please let me know.

I think you will find out if you have any AP users because there must be one that does not have a rudder sensor or something like that.

I have an idea of how must work it is to maintain Kip after I more or less from nothing compiled and tested kip on a system with live paths. I had to cheat by putting signalk in a container and hacking a problem with opencpn. You must have a working knowledge about every thing and the number of dependencies are frighting. 

I come from a different wold of programming than js and it been a while since I did any real programming.

I like to help just be aware I will be starting from scratch. That said When I look at the current public stylesheet it is only 0.5mb and I remember looking at the theme size once. A theme take up about 60% of the default theme.

If you still want me to look at the style sheet let me know. It looks like limited amount code and it would be good to know how it works. Please confirm we are talking about the stylesheet that is in the public directory.

PR #287 (2024-01-12): It is not possible to save wind widget with paths errors
Fixes #275. Added path group enabling based on widget Display Options prenventing saving of the form.
closed 2024-01-12 by godind