mxtommy/Kip #277: Add WakeLock option

godind, 2024-01-02:
Add a WakeLock option to prevent device from sleeping and locking the screen.

godind, 2024-03-16:
replace NoSleep so it works on non-mobile devices: https://medium.com/@dhashvir/how-to-setup-a-typescript-project-to-use-wake-lock-api-e2243512c012

JonyIvy, 2024-03-23:
The current implementation with NoSleep works on my Raspberry Pi with DSI touch display, if KIP is in fullscreen mode. In window mode, the screensaver kicks in as configured. For me this is quite ok.

godind, 2024-03-24:
Maybe not that important them. Just need to look at power and cpu usage of the current implementation.

godind, 2024-07-30:
Investigate nosleep library use case to determine if a more modern approach with less ressource consumption can be applied.

Feature requirement: in full screen, screen saver should not activate.

closed 2025-03-26 by godind