mxtommy/Kip #301: Datetime time zone issue

peugenio, 2024-02-14:
When displaying datetime in KIP, I selected the local timezone (Central US, i.e. "America/Chicago") which shows correctly.   But when I view KIP remotely in another time zone (Eastern US), the KIP display shows my local time (Eastern US) even though the setting clearly has another time zone selected (Central US).  This behavior is the same for both the Safari and Chrome browsers. I tried selecting a west cost time zone (Pacific US) and still the time reverts to local.   It appears that this is true for all time zones except for "UTC" which when selected shows the proper zulu time.

godind, 2024-02-15:
Hi. Thanks for the report!

Could you share how and where you set the origine time zone?

Note that the date/time conversion   Ruled uses the browser and OS configuration on the computer running KIP. Maybe the other computer, or the one running Signal K are not configured with the same time zones?

Think of it this way: if the SK server is configured as a computer  located in Japan, and its physically recording time in the US. The date will appear ok in SK and when running KIP on that server. But if you access KIP from another computer set to a different time zone then the SK box, the base date from SK will be from Japan, the computer will adjust to its OS time zone, then KIP will translate to the time zone you set in the widget.

godind, 2024-02-15:
We need to check all settings first and see where the problem is.

godind, 2024-02-15:
One way is to share the string SK sends.

peugenio, 2024-02-15:
Hi,

When adding the time field to KIP, I used the Widget Options field for “Adjust Value to Time Zone” and selected “America/Chicago”.    I then saved the configuration and load that configuration from another remote browser.  On the remote browser, the Widget Options field still lists “America/Chicago” but I am now located in Eastern Time which is what it defaults to.  

My fix was to just change the widget lable to “Local Time” rather than “Central Time”.   It not a problem, I thought I would point out the quirkiness of this.

Thank to all  for all you do.

godind, 2024-02-16:
Oh. You mean the default widget label when you add the widget?

If this is what you mean, what do you think would be more obvious to indicate it should be adjusted according to the users setup? Just Widget label?

I'm all for making things more obvious!