mxtommy/Kip #311: KIP crashes from time to time

Techstyleuk, 2024-02-25:
I have been running KIP on a new RPi 5 running Openplotter 4 Starting (12-29-2023), up to date.  Signal K is version 2.5.0, Kip is 2.7.  It is running in Chromium, and connected to the Signal K demo server.  

I am running it 24/7 and each time I come back to it, perhaps every 6 hrs, Kip looks like this:

[image]

Techstyleuk, 2024-02-25:
labels: bug

godind, 2024-02-25:
Hi,

Thanks for the report. Could you share the browser console log? That might help troubleshooting.

Also can you share a screenshot of the scream you lean it on. It could be a memory leak issue with a specific widget.

I'll see if I can find a problem.

Techstyleuk, 2024-02-25:
I think this is the console log entry associated:
[image]

I have not changed anything yet, so KIP is as per the demo:
[image]
this is page zero and was the one I was on

godind, 2024-02-25:
This warning image is just a meta key for Apple devices so you can save KIP as an App. It's needed even if Google complains about it!

I suggest leaving the console log open so you'll see what's log if the browser runs out of memory. I'll test on my and see if I can reproduce the problem but it could be related to your setup.

No one has reported a similar issue, yet!

godind, 2024-02-27:
Next Time it crashes can you check in the Signalk logs if you see any errors please. I ran for several hours with memory profiling but nothing unusual so far.

godind, 2024-02-28:
Can you also update Chromium and share the version please?

godind, 2024-03-21:
Is the issue still present? If so leave KIP running with the console log open and share the latest errors. 

Thanks!

Techstyleuk, 2024-03-24:
Sorry for my lack of feedback, I had a few other issues so last weekend I did a new install and I have been watching it, but not closely.  I did just see it crash but did not have the console open, I am now continuing to run with the console open.  I am also monitoring the memory usage, and it looks a little strange - this is all after the crash:
[image]
I was not doing anything that would cause the system to just increase memory, or to drop like that.

Techstyleuk, 2024-03-25:
I did reload KIP browser window when my memory was around 90%, as it wasn't responding and the Memory and CPU usage dropped dramatically: 
[image]

Techstyleuk, 2024-03-25:
KIP is at 2.9.0, I upgraded yesterday afternoon.  

Today I came home from work and it looks like this:
[image]
Chromium has crashed but is still on the taskbar, and there is still a black window there, but no tabs or anything.  I cannot get the log without restarting it.  when restarted it comes up normally.  I am running 4 tabs - KIP, Gmail, Signalk, Pandora Music.  judging by the graph below, I'd say it crashed at about 8:00am this morning:
[image]
Chromium version is: Version 120.0.6099.102 (Official Build) Built on Debian , running on Debian 12 (64-bit).

I am monitor the memory curves for a couple of hours then kill KIP and see if the curve flattens out

godind, 2024-03-25:
This will be difficult to find but with some patience and collaboration I'll fix it.

Can you upload your config file here so I can reproduce a similar layout? You sill find the file named 9.json under ~/.signalk/applicationData/<signed in user>/kip

Techstyleuk, 2024-03-26:
my KIP layout is just the demo and using the demo server for data.  I am running Openplotter 4, so Bookworm.  The only Kip directory was at: ~/.signalk/node_modules/@mxtommy/kip and the only json file is packages.json

godind, 2024-03-26:
Ok. So you are using local browser storage and not signed in to the server.

Running some tests on my end.

Techstyleuk, 2024-03-26:
I was signed into the server but am I signed into KIP?  not sure:
[image]
I killed the KIP window just after 8pm and the memory no longer escalates, so it looks like it is definitely KIP.  
[image]

Techstyleuk, 2024-03-26:
I noticed that when I went to KIP's Setting screen, the CPU usage and memory escalation stops:
[image]
I have since gone back to the earlier screen (page0) and the CPU and memory increase.  I navigated to page 1 (a big dial with COGS) and the CPU drops off and the Memory flattens

godind, 2024-03-26:
Nice find! So could be related to a specific widget.

Charts take up cup and memory but should stay flat.

Techstyleuk, 2024-03-27:
I just got home and it had crashed.  I had already deleted the Wind display and the historic True wind graph.  After waiting to see if the memory continued to rise, I deleted the "Apparent wind speed" widget (ref: the first picture on this thread).  after this was deleted, the memory stayed low, then I added back in the steering widget, and the memory continued to be flat:
[image]
Therefore it appears to be associated with the Apparent wind speed widget

godind, 2024-03-27:
Sorry, reading again I'm not sure about one point: 

In the picture, per yellow line sections, I'm not sure what widget was present or removed? You talk about Wind Steering, Data Chart (historical data) and linear gauge (Apparent Wind), right?

Techstyleuk, 2024-03-27:
I started removing stuff and the Historic data chart was removed first, then the Wind Steering dial and neither stopped the condition.  I then removed the Linear Gauge (Apparent wind) and this flattened the memory usage:
[image]
after that, I started adding gauges back to make sure the issue didn't come back.

godind, 2024-03-27:
Ok. So you did not add back Apparent Wind Speed widget and you did not confirm the leak comes back when it is present ?

I will test.

Techstyleuk, 2024-03-27:
Just got home and I am not sure it is fixed.  My Kip screen is frozen, and I had to reload it.  The graphs looked like this:
[image]
so it looks like it crashed, perhaps, at 8am.

godind, 2024-03-28:
Are you running in KIP fulll screen mode using the KIP fullscreen menu option?

Techstyleuk, 2024-03-28:
No, I run KIP in a browser window, occupying the right half of the screen, with OCPN on the left half of the screen

godind, 2024-03-28:
Just checking! I made a version with a few clean up for you to test. I am still not able to put a finger right on the problem. But I still have a few leads to investigate.

Can you install a beta version on your system to test please.To install using a command shell:
`cd ~/.signalk`
`npm install @mxtommy/kip@beta`

Once completed, hit the Restart link in the Signal K Admin site. Once restarted, in the Signal K Admin site's Appstore / Installed menu, you should see KIP 2.9.1-beta1 listed.

Run KIP and see if it's any better.

Thank you!

Techstyleuk, 2024-03-28:
it is running the Beta version as of now, I can run it overnight and then check it.  is there a quick way to reload to the default set of widgets if we want to use that?

godind, 2024-03-28:
Yes. Go in **Configuration  / Settings**, then **Storage** tab. At the bottom you will see a button labeled `Load Demo`

Techstyleuk, 2024-03-28:
OK, it is reset to demo and this is how the screen is set up:
[image]

Techstyleuk, 2024-03-28:
limited time in the morning, but after 8 hrs it is still running, but there still appears to be a memory leak:
[image]

Techstyleuk, 2024-03-28:
OK, at 6pm, this is what it looked like and KIP is frozen:
[image]
I will delete the linear gauge widget and check again in a few hours

godind, 2024-03-28:
I guess it will crash again. On tour next test, can you disable notifications and let it run? Go in **Configuration / Settings** - **General** and toggle Disable All Notifications option. You should see the notification menu button change icon with a disable sign.

Reload the page to clear memory and let run overnight.

godind, 2024-03-28:
It's been stable on my end for a while.

Techstyleuk, 2024-03-28:
OK, so I reset to default layout (as last night) but disabled all notifications.  I will let you know how it goes - thanks for all the help and your work on KIP!!

Techstyleuk, 2024-03-29:
this was this morning.  KIP is still working and hasn't frozen yet:
[image]

godind, 2024-03-29:
Ok. I'll look deeper into notifications.

Techstyleuk, 2024-03-30:
with Notifications turned off, KIP Crashed around 8am yesterday.
[image]
I'm deleting the linear gauge and then running it

godind, 2024-03-30:
I've been at it all week. Not easy to find as it takes many hours before crashing.

I'll try with only one blank widget and everything else's turned off.

godind, 2024-03-31:
I've ran KIP using Firefox for 48h with no problem and memory stays flat at 50%. The problem is Firefox have a bug with the radial gauge :(

I could not reproduce the problem with Safari either. That makes me think it might be related to how Chromium handles memory... I'll big this some more...

Techstyleuk, 2024-04-01:
Similar results in Firefox.  after running for 24hrs, memory is 45% and flat with all widgets in the default layout running

Aitonos, 2024-04-02:
I have been using KIP for long time and I have always found this issue.

 In Openplotter 3, it will remain solid as a rock if the browser is left minimized (but data is still flowing as the MIN/MAX keeps being updated). But if the browser or the app (Both versions)  are being maximized / showing the window, then it will collapse after some time ( hard to say or measure not a clear pattern) independently of whatever you do ( I have tried it with even 1 numerical widget).

I tried this with clean installation of OP 3 , and same issue.

I have tried with Mobile, tablet, Firefox, Opera, Chromium, etc and it always happens.

I learnt to live with this issue as it's a minor payoff compared for the good thing that Kip is!

Techstyleuk, 2024-04-02:
@Aitonos,
interesting, I have never seen KIP crash before in the 2 years I have been running Openplotter 2 and I always run the screen like this:
[image]
Never minimized, always half and half.  
with Chromium, I was seeing crashes every 8 hours.  I have been running for 48 hours (split screen) and the data for the last 24 is below:
[image]
there is some drift upwards of Memory, but much slower

godind, 2024-04-03:
Well that might be a good clue! Could be due to some threads going to sleep when browser is minimized , screen saver goes off or power management comes in... worth looking into.

Techstyleuk, 2024-04-04:
it finally crashed this evening after over 3 days of running on Firefox:
[image]

Techstyleuk, 2024-04-04:
I tried again with Chromium and it had failed in less than 24 hours and the Memory profile is clearly different, with Firefox looking very step like and Chromium building constantly:
[image]

godind, 2024-04-05:
@Techstyleuk You don't have the radial gauge spinning issue with Firefox?

Techstyleuk, 2024-04-05:
@godind if you are referring to https://github.com/mxtommy/Kip/issues/222 then no, I have never seen that in Chromium or Firefox

Sparhawk76, 2024-04-05:
I don't know if I am having the same issue or a different one. My use case is a bit different in that I run my OP3 install headless on a pi4-8gb.  I access kip via Chrome on my Windows based mini-pc.   I recently installed a second monitor connected to the mini-pc to have kip loaded all the time.  My config is 4 measuring radial dials at the top monitoring my batteries and solar setup, below them an embedded grafana graph of my battery/solar info over time, then below that 3 more measuring radial dials, and a text field (this row for starlink stats), then on the bottom row I have a radial baseplate compass (rudder position), and beside that an embeded Pypilot Control tab.  I have found that every morning since I set this up, I wake up and find the Chrome Aw Snap page.  A bit of searching later and I found this bug report,  which seemed current and similar issues.

godind, 2024-04-18:
Hey guys. SK 2.7.2 is out and fixes a leak that caused repeated connection termination on the server side. It takes a while before the bug starts. Like 6-8h. It did not crash the server, just killed all clients.

I'm not sure this will resolve the issue, but it's possible that unknowingly, after 6-8h, KIP's connection  is being repeatedly terminated causing an auto reconnect in KIP in the background.

Can someone test for me please? I dont't have a rip at hand.

Thanks

godind, 2024-04-18:
> eadless on a pi4-8gb. I access kip via Chrome on my Windows based mini-p

That means you get the problem in Chrome on windows, right? It's really hard to find this kind of problem! I'm rewriting lots of code and cleaning up all over. Have not found it yet. Next release will tell us more...

Techstyleuk, 2024-04-18:
this is this morning, after update to latest KIP and SK and a reboot:
[image]
Chromium window did not crash

godind, 2024-04-18:
Thanks for the report!

Have we won a small fight? Let see if need a few more fights to win the battle.

In any case, this issue has triggered a lot of rewrite and refactoring of KIP. All positive.

Techstyleuk, 2024-04-19:
when I got home from work today, KIP had crashed, looking at the curves, I cannot see the point where it failed, but it was working 12hrs ago.
[image]
[image]

godind, 2024-04-19:
So it doubles run time. Let's see if the next release improves things.

godind, 2024-05-06:
@Techstyleuk still observing partial improvements?

I'm still working on to publish a refactored version that should improve ressource usage and close a few "potential" leaks, although minor.

Techstyleuk, 2024-05-06:
I have been busy getting the boat ready to go back in the water, but I have this pi running in my Basement.  I looked at it and it was frozen, I reset it and then will check it in 6 hrs or so.

Techstyleuk, 2024-05-08:
it lasted about 12 hrs

godind, 2024-05-28:
Hi all. Latest release v2.10.0 has a few fixes. It runs better with n my end. Can you report when you find time. Thanks!

Techstyleuk, 2024-05-28:
I upgraded last night, It was still going strong after 12hrs. I had to shut it down for another reason, but will run it from later this morning and let you know in a couple of days

Techstyleuk, 2024-05-29:
It did just crash after 13hrs.  the funny thing is, this is my test environment running in my basement - Pi5, Openplotter V4 (Bookworm), latest KIP and SignalK.  I was on my boat all weekend (55hrs) and never turned the system off - it is a Pi4, Openplotter 2, an earlier version of SignalK and a 2 week old version of KIP, upgraded just before launch 2 wks ago

godind, 2024-05-29:
Thanks for the report. I can't pinpoint exactly where the problem is. I keep searching and plugging potential leaks. I double the time so far.

It is only reported with Chromium so far, not on all Chromium, os and hardware versions. It's related to Chromium install setting or builds and/or the way it manages the Java virtual machine memory cleanup and memory allocation.

godind, 2024-06-16:
@Techstyleuk is KIP still running fine on pi4 and crashing on pi5 after 12+ hours?

Just keeping track.

Thanks

[4 more replies]

closed 2025-05-28 by godind