mxtommy/Kip #312: Disable automatic saving in Default

ignamendi21, 2024-02-26:
Hello,

I am using KIP connected to my signalk server, 
I have created my windows and my perfect configuration of data visualization, however sometimes when someelse uses my account and star meshing and playin , then I return I I find all my screen broken and updated....
I know that we coul have differtent users etc... but I would to disable the automatic saving in Default that KIP does all the time...

Is it posbile?

I want everything as it is right now, but disabling the automatic saving in Default. I know that when I create custom windows I will have to save it in a remote configuration and then I will have to copy it inside Default. But I will do it once and that is. 

How can I disable automatic saving in Default?

godind, 2024-02-26:
Hi,

This feature is not available at the moment. As you stated, to work  around this create a separate user for each. You can copie a user's default to Global them from global to the other user. I also suggest making a backup to Global.

godind, 2024-02-27:
Thinking about it I think the easiest way would be to give the SK user Read Only permission and switch it to R/W when required. Not sure if that will generate unforgiving errors when you play with config but worth a try. We might only need to add a small feature that handles that is case.

Otherwise the app stores config as changes are made from different part of the app. All changes go to the Settings service which, depending if the Sign in is enabled, can either be configured to use the local browser storage or the Storage service to store config.

ignamendi21, 2024-02-27:
mmmmm I think it works.
I create a Read/Write user, then I create all the windows and screen I need. Then from an Admin user, I change that user to only Read permisison, and now it doent work the aoutosaving in Default. So if someone meshs it, only have to reload the browser and the initial configuration I left will come out again.

closed 2024-02-27 by godind