mxtommy/Kip #318: SignalK Authentication Infinite Loop

jtroyer76, 2024-03-06:
Kip can get into an infinite loop when signalk authorization token is set to never expire in security settings.

[image]

Here is a snip from the console log:
[image]

The issue is KIP tries to decode the expiration date of the token which is not set.

A workaround is to set the expiration to 10d or any other valid value.

godind, 2024-03-11:
It is cause by the NEVER value in the Signal K / Security / Remember ME Timeout parameter.

Change the parameter to "1d" or another value for a temporary fix.

instantchow, 2024-03-11:
Is this out a problem with the SK + KIP "out of the box" I think I hit this issue/or something very similar when I initially attempted to turn on Authentication (to get token and use remote storage)

In my instance:

`/admin/#/security/settings` 

Doesn't show "NEVER" it was simply blank.

godind, 2024-03-12:
As of sk v2.6 it is the default parameter value. Its recent issue that Kip needs a fix for.

Not sure why it's blank... I think the UI does not allow NEVER but in the config file its probably set to never. Put 1d in SK, save and restart. It should fix it.

PR #327 (2024-03-18): Login/Logout Authentication Infinite Loop
Added NEVER expire token support. Fixes #318
closed 2024-03-18 by godind