mxtommy/Kip #319: Backup for Kip using the UI within any browser

x7y8z9, 2024-03-07:
After loosing my Kip instruments & lots of time I invested because of reinstall of SignalK on a Victron CerboGX after OS update, I do request a feature to perform a backup using the UI: 

One button at Settings/Storage/ConfigManagement to download the Kip settings (instrument panels, Zones, Historical Data and units) using the browser of any PC/Android/IOS device.
One button to restore the Kip settings (instrument panels, Zones, Historical Data and units).

godind, 2024-03-08:
Just as info in case you are not aware: if you use the Sign in option on all devices and use the same user, the config is automatically retrieved from the server.

x7y8z9, 2024-03-08:
Yes correct, that's the way every device is getting it's instruments onboard of our installation.

I'm asking for a simple way to bring Kip back to live after a factory reset of either the Victron Cerbo (Victron OS plus SignalK) or SignalK. Same situation in case the Raspi needs to be replaced by any reason.

Please, try to find a solution to 1.) pack the files defining a individual Kip-installation 2.) download such a backup 3.) restore such a installation.

godind, 2024-03-21:
KIP configuration is included in the Signal K backup/Restore feature part of the Admin UI. It is recommended to do a backup before any upgrade and replacement with new hardware so that you do not loose it and can easily restore.

This should be sufficient.
Thank you

closed 2024-03-22 by godind

x7y8z9, 2024-03-22:
I do not agree.

A backup of SignalK produces min. 50MB of data and requires a full restore of SignalK. There is no option to restore just Kip.
This is no solution for the need of a backup to store the Kip definitions.

Plus it doesn't follow a use case like: Victron is updating their OS together with a new version of SK. 
A backup of a outdated SK-version including Kip plus several other plugins won't be a practical solution for a Kip-restore.

Let's compare your suggestion with a backup of a complete PC, the OS and any installed software for to get a backup of a database. 
This is simply not a smart or applicable solution.

Please check for the option for performing a backup of Kip.