mxtommy/Kip #331: Setting & pages lost after updating 2.7 -> 2.8

x7y8z9, 2024-03-19:
I received a notification for update to 2.8 within SignalK. After performing the update the (still open) default pages (F5) on a W11 PC ended in a reload loop. Closing and cleaning the browser cache loaded an empty Kip page. Sign in and copying user/default or global/default from remote to local do not load any predefined pages and settings.

Hopefully 2 weeks of defining are not gone...

godind, 2024-03-19:
If you have a global backup you should be fine to restore later.

What version of SK are you running?

If it's a bug I'll need to fix it quickly!!!

godind, 2024-03-19:
Also not that if you run KIP with sign in option, local config has no effect. It uses what's in the user storage: user/default.

Keep your backup in global/default and don't overwrite it manually! KIP will not touch global on its own, unless you manually copy over it.

x7y8z9, 2024-03-19:
SK 2.50 on Victron CerboGX OS 3.22

x7y8z9, 2024-03-19:
btw. that's why I made a feature request for UI-based creating updates & downloading everthing Kip needs.

godind, 2024-03-19:
Ok. V2.5. It should not a bug that is related to 2.6.0. Just checking.

So to recap so I can nail this right. You have:

- one Kip 2.8 running in sign in mode
- one Kip 2.7 running in sign in mode
- both connected to the same sk 2.5
- both Kip use the same user to sign in

When you signed in with v2.8 it started looping?

What the state of v2.7?

You probably 2.7 should flush use/default and start from blank, if you restart it, as it will not know about v2.8 config.

If you don't restart v2.7, v2.8 should have upgraded the config and run fine. Is this the case?

godind, 2024-03-19:
> btw. that's why I made a feature request for UI-based creating updates & downloading everthing Kip needs.

The request for backup is really a Signal K config backup need as the KIP config is held by SK and you probably want to restore the full SK configuration, including app config data (KIP, Freeboard-SK, etc).

That said, with KIP, you can always Copy `Remote - user/default` to `global/"what ever label"` to make a backup copy - recommended before any KIP upgrade.

You can also save KIP config locally and later move it to an upgraded server or a new/different SK server: 

1. Copy `Remote - user/default` to `Local Storage`. This will copy to the browser's local storage. 
> Note that Local Storage config is only used by KIP if you do not use the sign in option. Whitout the sign in option, everything is in Local Storage and independent in each browser.

2. Connect to a different server (or that same server after upgrading Victron) and sign in with a user. Because this server and user does not have a config, KIP will create a new blank config.
3. Restore your saved config to a new server by using: Copy `Local Storage` to `Remote : user/default`

x7y8z9, 2024-03-19:
Scenario to get into the infinite loop after update

Kip 2.7 was running on a (Navigation) Intel based, Gen. 13 PC, W11, in the Edge browser session/Kip installed as app.
On a W11 laptop I started SignalK Dashboard using FF123.0.1. SignalK showed an available update for Kip and I chose install. 
I used the same admin user on both sessions.
As soon as the update has been finished on the laptop I closed the Edge-session on the Nav PC and restarted the Kip session.
This ended in a infinite loop on the Nav PC.

I started a Kip session at the laptop and I got a empty i.e. clean Kip session without any of my pages. I did a login, nothing changed.
I performed a copy from Remote to local of User/default: empty pages
I performed a copy from Remote to local of Global/default: empty pages

I did a restart of SignalK: no changes
I did a restart of Victron CerboGX: no changes

godind, 2024-03-19:
So both user the same logged in user (admin) on the same SK server. In this case they both load their config from the server at user/default and not from local storage.

If you have a good v2.7 copy of the config in global/default. Have just one instance of Kip. Sing in with admin and copy remote global/default to remote user/default. KIP should upgrade it and restart. Then start the other KIP with the same admin user and you should be good. Let me know. I'll fix any bug.

x7y8z9, 2024-03-19:
Ok, without sign in: copying from remote global/default to local shows the last working definitions.
What’s the next step?

godind, 2024-03-19:
So if the global/default works fine as a local storage config.

Sign in with the admin user. I'm not sure what state the user/default config is in and what KIP will do but if you can get to the storage tab, copy local storage to user/default and it should reload and work.

If it does, make a backup from remote user/default to global/version28 so that you have a good backup. Once all is fine you can eventually delete global/default. Anything in global is only for storage - safe keeping. 

Let me know

x7y8z9, 2024-03-19:
Got it, Kip is back to life.
Thx for supporting!

Next step will be updating the CerboGX to Victron OS V3.30.
I worry about losing SK and Kip settings in case Victron updates SK to 2.6x.
We do use a vessel management system which contains backups of every IT/IoT-device on board (on NAS plus cloud storage).
Kip is the one-and-only software why SK is installed at the CerboGX and the only piece of software which can't produce any (traditional) backups.

Joerg

godind, 2024-03-19:
Why don't you simply Backup/Restore the Signal K config? It will grab KIP and everything else. It's a simple matter. See in Signal K admin IU under Server - Backup/Restore.

You can download with just a click!

I'll close this issue.

closed 2024-03-19 by godind