mxtommy/Kip #348: Zones displayed incorrectly if units don't match

tvr256, 2024-03-25:
Zones should be converted from the zone units to the gauge units before displaying on the gauge.

Example:

For a SignalK field with units "ratio", set up the following zones with units "ratio":

[image]

Create a radial gauge with units "ratio", the zones are displayed correctly:

[image]

Change the gauge units from "ratio" to "Percentage value". The zone alarms still trigger correctly, but the zones are not displayed correctly on screen. The zones should be converted from the zone units to the gauge units before displaying.

[image]

tvr256, 2024-04-06:
Hi @godind , this fix doesn't seem to work for me. Gauges with larger numbers work fine, but a ratio (0-1) gauge doesn't display any zone information now:

[image]

[image]

godind, 2024-04-06:
Grr... it's that bug where fractional numbers such as 1/3 can't be represented properly in JavaScript. Same as the ticks numbers. I'll fix it in the next release!

godind, 2024-04-15:
Coed in next release

PR #369 (2024-04-05): fix: radial and inear gauges zones not drawn to proper units
Fixes #348: Zones area drawing did not support path format conversion.
commit 28978b7a (2024-04-05): Zones displayed incorrectly if zone and gauge units don't match
Fixes #348
closed 2024-04-15 by godind