mxtommy/Kip #350: Ideas about new data chart.

petsicola, 2024-03-26:
I think that the new data chart is a very good thing.  I'm going to purpouse some ideas.

1º vertical values. would be a good idea  put the vertical charts

[image]

2º The value in the center. there is an option to put the center line in the midle with the value. 
[image]

3º The font size of actual values could be big?

4º multiple values in the same chart? I don't know if this is interesting for anybody.

Question about moving average. what happens with you mooving average if you put true wind values of 5º and 355º give averages of 180º??

Mxtommy kip is practically perfect! thank you for your wrork.

godind, 2024-03-26:
Hi. Thank you for the feedback. This was step one for getting to what you asked in the other issue 😉

Angle average calculation for degrees is not done. Kip works in Degrees by default and I need to check why it's not in base radian.

godind, 2024-03-26:
I have done Exponential Moving Average (EMA) and Double EMA calculations but not on angles, Angles should use calculations from link found in the Issue with tang2().

godind, 2024-03-26:
I had to get to a few bugs and it was good as a v1 to get feedback!

godind, 2024-03-26:
If you want to PR some changes you are more than welcomed!

To keep widgets somewhat easy to use for users, their benefits accessible and maintain in code. We will need different ones. The Data Chart is about visual trends over time and so the values are not that critical, hence the smaller size.

For racing/sailing charts, I think we need different widgets with their own specific layouts and config: vertical with larger numbers, multiple series, polar data maybe, different scales, custom settings, etc. I would do new widgets for that purpose. What do you think?

petsicola, 2024-03-26:
I have this ideas,

[image]

godind, 2024-03-26:
Are you planning to add new widgets?

godind, 2024-03-26:
Note that EMA and DEMA are not yet exposed as I need to add angles average. Too much stuff to do!

petsicola, 2024-03-26:
I'm not programmer, I'm trying your widget. i only can make that for you. and if I say that you can cretate a data chart widget the minimum is try the widget and look the maximum posibilities with this widget.

yes, with this charts we only have the problem with the wind direction magnetic.... and only with values close to 0 or 360.

I'm thinking in things as:
- Chart of open cpn, are the right side charts. polar performance and SOG
[image]

- a chart with VMG and porlar VMG performance
[image]

petsicola, 2024-03-27:
> Note that EMA and DEMA are not yet exposed as I need to add angles average. Too much stuff to do!

I'm aware that EMA and DEMA haven't been revealed yet, but I believe it's important to understand the impact of turning the Moving Average on or off. This can be observed in the chart you've made. The differences are around 30 degrees, but with the Moving Average enabled, it essentially becomes a straight line.

[image]

godind, 2024-03-27:
Are you on Discord? I found user petsi but not certain it is you.

I'm no math expert and any help or advice would be great. If you are Discord, after I'm done with a complex bug, maybe we can setup a live call so we can talk and I can better understand?

godind, 2024-03-30:
I'm putting out an improved average for directions.

petsicola, 2024-03-30:
Working with the moving average in Node-RED, I realized that different situations require a different number of values in the moving average. For example, if you want to know stable wind values, you can include a high number of values in the moving average, like 40 values. If what you want to know is the apparent wind angle, it's fine to make the average with the last 10 values. If you want to know the current performance, you use relatively small average values, like 10. If you want to know how performance varies over time, for example with a new trim, you can use higher values, like 40. By high or low values, I mean the number of elements you will take into account in the moving average. Higher values generate more stable values but with a lot of delay to changes, smaller values generate less delay but the values are more unstables. 
I want say with this, that put the values that you want assign is important.

petsicola, 2024-03-30:
To implement your strategy effectively, you would:

Implement the moving average to calculate average values for the data chart. This involves selecting a suitable number of data points for each average calculation, based on the specific requirements of the data you're analyzing. For more stable values, such as average wind speed over a longer period, you might choose a larger number of data points. For more responsive values, like sudden changes in wind direction, you might choose fewer data points.

Create a plugin to Apply the moving average directly to the environment.wind.angleApparent data path. Since many derived data values depend on the apparent wind angle, stabilizing this value using a moving average can significantly improve the reliability and usefulness of the calculated derived data path values. By smoothing out the fluctuations in apparent wind angle, you ensure that the derived data, which might include calculations related to sail settings, navigation, or performance metrics, are based on more consistent and reliable input data.

petsicola, 2024-03-30:
A dataset gives that. something is bad...
[image]
This appears ok I put in the display paths the paths that I'm testing.
[image]

petsicola, 2024-03-30:
and here the problem that says. a value of -170 and +170  is not 0
[image]

godind, 2024-03-30:
It because this widget does not support rolling (wrap-around) scales. You'll not get what you want with this widget. You can check the calculations if you want.

As for data, when working with time scales, all the data has to be in sync for both time and frequency to create stats and data points, on time points. The way plugins are built at the moment, it's not really possible or easy. They all come at different times and different rhythm. Try and plot multiple paths and create an average at a specific regular time  interval. You'll see the challenge.

I think we need a performance plugin that does that and sends data in sets.

godind, 2024-03-30:
> Working with the moving average in Node-RED, I realized that different situations require a different number of values in the moving average. For example, if you want to know stable wind values, you can include a high number of values in the moving average, like 40 values. If what you want to know is the apparent wind angle, it's fine to make the average with the last 10 values. If you want to know the current performance, you use relatively small average values, like 10. If you want to know how performance varies over time, for example with a new trim, you can use higher values, like 40. By high or low values, I mean the number of elements you will take into account in the moving average. Higher values generate more stable values but with a lot of delay to changes, smaller values generate less delay but the values are more unstables. 
> 
> I want say with this, that put the values that you want assign is important.
> 
> the data set make that... I think...
> 
> 
> 
> I'm analazing a file with mxtommy kip and When you tack in the boat, the values of the data chart take about 10 seconds to detect that you are making the tack, more or less. 
> 
> 

Yes. That's what a simple moving average (SMA) is. Data Chart uses a 0.25 ratio of data points. We need different widgets for different purpose. Remember Kip has different goals then Grafana. We just need to have what sailors need to see, presented in a "marine" format. So if we explore outside that scope, we'll find gaps. Grafana has its place there and can be integrated into Kip. Just saying...