mxtommy/Kip #363: widget label

jimmyendt, 2024-04-03:
Hello, 
I'm fairly new to using Signal K and KIP, but really loving it and all the hard work! But i have to admit I have no experience coding or anything. 

I was wondering, would it be possible to make the Widget label removable? I know you can leave it blank, but I have the feeling that it does take up some space in the awesome boxes you can put the widgets in? is this correct? 

it would be amazing to maximize space maybe? 

i'm also on discord and maybe have some more ideas/questions if that is okay.

thanks in advance

godind, 2024-04-03:
Hi. Thanks for the feedback. It's possible but would require reworking all the Widgets. I am not convinced this use case would apply to most. It's would make pages hard to understand without any label. All marine package have labels for that purpose.

jimmyendt, 2024-04-03:
Thanks for the fast reply. Sounds like a lot of work and it's merely a question.
I was thinking about the wind-widget, or radial gauge widgets that would speak for itself and therefore maximizing the space. 

can i ask you another question on discord maybe?

godind, 2024-04-03:
Absolutely!