mxtommy/Kip #386: Side scroller appearing on interface, no way to remove

markpires123, 2024-04-18:
Since I updated to the latest version, I've noticed all my browsers show a side scroller on the right. I can just barely nudge the page up and down with this. I reset my configs, deleted them, created a new one but still see this side scroller and it affect the overall layout of my previous pages.

markpires123, 2024-04-18:
I dunno if its the latest version with openplotter,  but I was able to fix it but adding overflow:hidden to the css in the root of public directory. styles.6af848a7c959e550.css  I added it to the body of app-root.

godind, 2024-04-19:
Fixed in next version by making navbar buttons larger to accomplish for larger touch area.

godind, 2024-05-03:
fixed in #385

closed 2024-05-03 by godind