mxtommy/Kip #392: Baseplate compass bug

voilier-kalymnos, 2024-04-25:
In baseplate compass mode, negative value don't work.
The range is 0 to 360°, in my case the wind sensor send me data from -180° to 180° 
[image]

godind, 2024-04-25:
The compas gauge only support path with positive rad values. Add gauge feature to support for paths rad unit that use negative values (-180 to 180 in degrees).

godind, 2024-04-29:
@voilier-kalymnos  Not as easy I as thought. I think there might be another path the converts this. Maybe with derived-data. Ask the group again. Else I'll create a dedicated Widget. The marine compass broken as you know...

PR #413 (2024-05-23): feat: add compas support for Rad units with negative to port
Fixes #392
closed 2024-05-23 by godind