mxtommy/Kip #398: Demo display not persistant from demo to signed in mode

Ribodo, 2024-05-02:
Since the upgrade I have been unable to get the "demo" layout to show once I log into the server as it use to with older versions. I like to use the demo layout and modify it as the starting point.

I go into the setup once logged in and click on the demo button, I would like to save this setup but am not able to do this. Upon connecting back to my local host the demo layout vanishes and I am left with a blank screen, having to start from scratch on the layout.

Ribodo, 2024-05-02:
I have been playing with it and it only seems to cause an issue if I put the connection into Signal K Authentication mode. without this enabled I seem to have the Demo display visible

godind, 2024-05-02:
When you load demo settings, it resets the layout, and connection to demo server url without sign-in option. Without sign-in, the config goes to local storage.

When you later enable change the connection and enable sign-in, it takes the config from the sever, which does both contain the demo layout.

What you need to do is copy the local storage config to that user.

Sign-in and go in Configuration - Settings, in the Storage tab, copy local storage to user/default. That will fix it until this bug is resolved.

closed 2024-05-03 by godind

Ribodo, 2024-05-03:
Thanks David. I'll try it later

PR #399 (2024-05-03): Demo display not persistant from demo to signed in mode
Fixes #398 Demo config not loading with sign-in connection option.
commit 57467a89 (2024-05-03): Demo display not persistant from demo to signed in mode (#399)

Fixes #398