mxtommy/Kip #408: Paths with null values

Fgonzalez22, 2024-05-18:
Im trying to set a gauge with COG from GPS
As the boat is stationary at port Signal K report Null, and in KIP Data, it shows up with no data.
I guess thats why I cannot create the gauge, KIP doesnt recognise the path and wont save it.

Any suggestions?

godind, 2024-05-19:
If the path exists, you should be able to save it, regardless of the data it sends.

Can you see the path in SK? Maybe SK does push it?

You are getting COG from derived data?

Fgonzalez22, 2024-05-19:
Yes i can see the paths both in signal k (null) and in kip(no value). COG both GPS and derived data.
Attached pictures in case the are usefull.
[KIP.zip](https://github.com/mxtommy/Kip/files/15369892/KIP.zip)

godind, 2024-05-19:
It looks fine. I think you simply need to select a source and format in KIP. Maybe something wrong with the source emitting the value. 

KIP automatically selects the source if only one is present. This indicates that you have more than one source emitting for the same path. You should either configure SK source priority so it's handled by SK. Or force a source in KIP.

Fgonzalez22, 2024-05-19:
I have tried with self.navigation.courseOverGroundTrue which i have only one source. I can see it in KIP data without value, but when i copy it to the path i get the message: Path does not match any known valid path. And neither before or after i can select source or format nor save it. The fields are grayed.

I have tried other paths without values and the same thing. I can only select those with actual values

godind, 2024-05-19:
Is the path listed in KIP's dropdown list of paths?

godind, 2024-05-19:
I can see from your screenshot that you have derived-data and gps.GP as sources for that path.

In KIP, can you see any sources for that path ?

Fgonzalez22, 2024-05-19:
> Is the path listed in KIP's dropdown list of paths?

No I cannot see it there, only tab

Fgonzalez22, 2024-05-19:
> I can see from your screenshot that you have derived-data and gps.GP as sources for that path.
> 
> In KIP, can you see any sources for that path in the widget options?

In the widget options i cannot open the menu because it does not acept the path, but in the data browser I can see the path and both sources.

Anyway, even if i have one source without value, like self.navigation.courseOverGroundReal, I cannot get KIP to acept the path even it being with one source and shown in data browser

godind, 2024-05-19:
I would need to debug this. Just looking at the code, null should not prevent a path or source from being listed and configured.

Something's wrong somewhere with null value.

If you can get it to emit a value other than null it will probably start to work.

Fgonzalez22, 2024-05-19:
> I would need to debug this. Just looking at the code, null should not prevent a path or source from being listed and configured.
> 
> Something's wrong somewhere with null value.
> 
> If you can get it to emit a value other than null it will probably start to work.

I dowloaded the delt simulator for signal k and once i got a value running for the path everthing when fine. I guess that to setup paths at the moment values are needed.

Thanks for your help !

godind, 2024-05-19:
Maybe, but if it's the case it's a bug.

I'm just not sure what SK sends over but the fact that the KIP data browser shows a source means that something is sent to KIP. I'll need to troubleshoot further.

godind, 2024-06-16:
We have had a similar issue. It turn out not all null values are published to paths. It depends on the provider being used. If the null values are not published, and no other values ever got published, KIP will not receive a delta update and so not list the path.

Closing this bug as it's a SK provider issue.

Thanks.

godind, 2024-06-21:
The source of the problem has been found.

Issue: Null values on the very first value delta make KIP tag the path type and Object, preventing widget that filter on data types from listing the path.

Fix: look at meta Units and if defined, force path type as numeric.

commit a509b0f6 (2024-06-22): Godind/issue408 (#432)

Fixes #408 - Numeric paths with null values not listed in Widget Options
closed 2024-06-22 by godind