mxtommy/Kip #420: Values jumping between sources after update 2.10.0

x7y8z9, 2024-05-29:
When there are different sources for the identical path the values within the instrument panel are jumping to another source.
The values are not jumping within SK or KIP data browser.

SK data
https://github.com/mxtommy/Kip/assets/143433271/c08d1a0e-2e79-4ada-93de-83dfea9f13af

KIP data
https://github.com/mxtommy/Kip/assets/143433271/e11e3583-5c4a-4816-8f76-fcd7ace9be2c

KIP instrument
https://github.com/mxtommy/Kip/assets/143433271/d6e4f801-3400-4ac2-b952-7fb9911b90ce

godind, 2024-05-29:
Did you notice this problem in prior versions?

If you have multiple sources for the same path, Source Priority for each path needs to be configured in SK, else widgets will jump between sources as they publish values. This is as per specs.

Check this and report back.

Thank you

x7y8z9, 2024-05-29:
I do use KIP for exactly one year. No prior version had jumping values.

Simplest example: engine hours of port and starboard engine, both displays/value do jump now (as well as RPM, fuel rate, load, voltages, amps, temps...)
Let's say: all configured instrument pages do have lot's of jumping values.

Kip is not usable any more since updating to 2.10.0

Pls check!

x7y8z9, 2024-05-29:
I do not agree on this logic:
The attached screenshot shows ONE selected path in the field "Signal K path" and ONE selected source from multiple sources in the field "Data source".
In my understanding this is a 1:1 definition and not 1:any (1:the last as of my observation)

Pls check the screenshot

godind, 2024-05-30:
I see 8 different sources for path electrical.batteries.0.voltage

They appear to be for different batteries. If it's the case, you should configure one path for each battery instead of multiple sources for the same device (path). 
- electrical.batteries.0.voltage
- electrical.batteries.1.voltage
- electrical.batteries.2.voltage
Etc.

I'm not saying there is no bug yet, but it's not good practice. Multiple Sources are meant to be used as backup providers for the same path. Line if you have 2 gps.  Use gps-1 and if it's stops sending data or breaks down, use gps-2 so the path still gets the info. 

Can you share a screenshot of the volt gauge configuration please.

x7y8z9, 2024-05-30:
Yes, there are multiple sources on lots of paths, not just for electrical.batteries.0.voltage.
For the two engines it's the same method.

And the paths and sources are not defined by myself: I'm using SK on a Victron CerboGX. So the paths and sources are predefined and no changes can be done as every update from Victron will overwrite it.

Once again: no problem up to and incl. V2.9.3.

godind, 2024-05-30:
I'll run checks on multiple sources.

Until I find the problem, please ask in the Server and Victron Discord channel how to change the setup. It's not how sources are intended to be used. Even if it worked before. You are at risk of having other problems down the road.

It will probably be a much faster fix. I have 2 engines and 5 batteries myself and they all use single paths so it's something in the setup, or the plugins that needs a change.

godind, 2024-05-30:
You can also down grade to the previous version until the fix. Uninstall KIP, restart the server and from the command line in the .signalk folder run: npm i @mxtommy/kip@2.9.3

godind, 2024-05-30:
Here is what I got from Discord:

> Best solution is to filter out the n2k data and use the Venus plugin.
> Or those instance numbers can be changed on the GX device. Instructions on the Victron site I believe.
> There’s no reason to bring that data into sk Since the Venus plugin brings that in and more.
> Filters on the Data Connections to ignore Venus n2k data.
> [image]
> I actually disable the n2k out on the GX. And use sk-to-n2k to send only what I need out on n2k.

See:
https://www.victronenergy.com/live/ve.can:changing_nmea2000_instances

godind, 2024-05-30:
I'm testing and I don't see source value jumps on my end. I've asked other user to check with their setup. Let's see if they report issues. That said, I strongly recommend you use the instructions above to clean up your setup and use individual paths. Go on Discord in the server channel to get support from the community.

x7y8z9, 2024-05-30:
Sorry, I don't get it: up to v2.9.3 KIP followed the rules/was able to run properly using Victron CerboGX and SK as predefined by Victron. Venus Plugin has been automatically installed and predefined by Victron large OS.

The data structure of using "paths" and substructure "sources" does make sense to me.

After installation of 2.10.0 not one of my instrument pages are in working order. On each instrument panel multiple values are jumping between sources within each path. This doesn't make sense to me.

I don't have to move the vessel until Monday morning and I'm actually not able to...

Please have a look at your code and the changes conc. reading data structures! (without the mentioned work-around of using another plugin)

image.png (view on web)<https://github.com/mxtommy/Kip/assets/34038713/66822464-e271-47d3-9148-f36123256b9b>

https://www.victronenergy.com/live/ve.can:changing_nmea2000_instances

godind, 2024-05-31:
Wow! I've been patient and polite... I think I need to set something straight with you. KIP is a software I build for myself, on my one personal time, for my personal pleasure.

You don't appear to appreciate something that you choose to use, for free, or the free assistance that you get either. Your directive tone is not appreciated. 

Use another software.

tvr256, 2024-05-31:
I understand David's annoyance, and I really appreciate everything he does to develop an amazing tool and share it with us.

Having said that, I'm also seeing the same issue with the latest release.

I have multiple sources for the same data, and I've configured KIP to display the data from a particular source. It has worked perfectly for a long time, but the latest KIP release normally shows the wrong source, and sometimes jumps rapidly between the two sources:

[image]

[image]

[image]

@godind , is there anything I can do to help debug?

godind, 2024-05-31:
@tvr256 closing this issue is not about having a bug or not - some people just have no manners or appreciation. It's something no one has to suffer or tolerate.

As you can read in this thread, I've been troubleshooting and talking to others about proper setup to help his configuration that, as I mentioned, regardless if there is a problem or not (meaning until I find the problem),  reds to be changed.

Sadly not every report is a KIP bug and so I first need to pin point the issue so I can figure out a fix. In another issue he had opened, I spent 3h investigating to find out he had not configured the Scale Start/End properly. That's fine, it happens.

The challenge is I can't reproduce it with my setup. I also asked others in the #kip channel to test. Mat says he can't reproduce either.

May I ask if you join the conversation so we can try to figure out where the problem is coming from as a group? I'm hard at work with boat launch preparation so any help will speed up the resolution.

Thanks

x7y8z9, 2024-05-31:
Dear fellow fighters for good software,
I would like to make it clear that I do not know David as a person. As a result, I cannot and do not wish to criticise him personally.

My comments related to a piece of software for which I was and am prepared to invest my time. Deriving, recreating and documenting malfunctions or missing functions is not intended to criticise the developer, but to improve the software - for my benefit, of course, but also for the benefit of all users.
I would like to apologise if my previous comments were misleading or could even be taken as a personal attack! This was never my intention.

At the same time, however, I would like to call on you to work together on optimising the software in an objective manner and with a forward-looking approach. Emotions are never helpful in this process.
(Including on my part, even if none of my instruments work a second time after an update).

Cheers Joerg

Side note: English is not my native language and I am located in Europe. And yes, I partly use translation software.

image.png (view on web)<https://github.com/mxtommy/Kip/assets/117984987/38b4cd98-434a-426e-9ff2-96b7a7f1af4c>

image.png (view on web)<https://github.com/mxtommy/Kip/assets/117984987/c908d8c2-f2f5-40f3-bdaf-c0e744e9c521>

image.png (view on web)<https://github.com/mxtommy/Kip/assets/117984987/f86ef9a0-9b1b-4c99-9d85-b133bb16b09b>

godind, 2024-06-01:
User Mat has found that using source Default or the first available source in the list is fine. The problem is only visible on sources past the first one.

Possible workarounds until the fix are:
- Rollback to previous version. On the server, uninstall KIP, restart Signal K server and from the command line, go to the .signalk folder (depending on your setup this can be `cd ~/.signalk`, execute command: `npm i @mxtommy/kip@2.9.3`
- Configure SK Source Priority, if it makes sense. 
- See if you can remap sources to separate paths, if it makes senses.

closed 2024-06-04 by godind

x7y8z9, 2024-06-05:
I confirm: with V2.10.01 multiple sources of one path are working again.
Thx!

PR #422 (2024-06-04): Update 2.10.0: values jumping between sources
Fixes #420
commit cbb2df91 (2024-06-04): Values jumping between sources (#422)

Fixes #420