mxtommy/Kip #421: Update 2.10.0: values not availabe on instrument

x7y8z9, 2024-05-29:
A path/source is availabe in SK as well in KIP data browser and KIP instrument.
But the value of the path/source is available in SK and KIP data browser but not shown at the instrument.

Note: setting the path filter to all shows the path/source, after confirming the filter shows Voltage but not power (electrical.grid.31.power)

SK
[image]

KIP data browser
[image]

KIP settings
[image]

KIP settings filter
[image]

KIP instrument
[image]

godind, 2024-05-29:
The path is of type Power with base Units of Watts (W).

Questions:
1- with the filter on all you see the path, right?
2- with the filter to Power (W) do you see the path?
3- with the filter to Voltage (V), you should not see the path. Do you see it?

In your screenshot I see the path  Format as Potential (volts). Is this something you manually set or was it auto populated? Watts source value to volts conversion will not work.

x7y8z9, 2024-05-29:
See below...

1- with the filter on all you see the path, right? Yes
2- with the filter to Power (W) do you see the path? Yes
3- with the filter to Voltage (V), you should not see the path. Do you see it? No

[After I saved the path and I open the settings window again it has changed the filter to voltage]

It's automatically Watt, which is correct for "power". See screenshot.

godind, 2024-05-30:
I don't see anything below. Maybe the send did not work.

What is the Scale Start and Scale End in the Display options tab?

You are saying that once you select the path, it automatically changes the filter to Voltage and the format to Volts?

x7y8z9, 2024-05-30:
I put the "see below" into a screenshot attached.

Correct: "You are saying that once you select the path, it automatically changes the filter to Voltage and the format to Volts?"

godind, 2024-05-30:
I'll investigate why. From some reason I do not any screenshot after the original post.

godind, 2024-05-30:
The filter options only serves to filter the path dropdown list when you type to search for a path. Say you want a path of type power: set the filter and start typing. Note that not all paths have proper Units defined so you can use All and Unitless to look for those. Once the path, source and format is stacked and saved, the filter has no effect on the value the widget will produce. When you re-open the widget config, it will be set to Voltage again as it is the default filter for this widget.

I have tested by setting a path Units to W and I do get data. I see that your value is -165.3 and -172. Check that the Display tab options Scale Start/End allow for this range. The default is from 0 to 15.

x7y8z9, 2024-05-30:
My bad, sorry: scale wasn't defined correct (0 to 5000 instead of -5000 to 0)

Thx!

closed 2024-05-31 by godind