mxtommy/Kip #425: Something weird with pressure units in KIP

ESchlotfeldt, 2024-06-09:
I have SK data source emitting an oil pressure reading of 4500 hPa, while the dial in KIP is displaying 45 hPa. 

Any thoughts? 

Thanks!

godind, 2024-06-09:
Hi, 
I looked at the code quickly and the Pa to hPa conversion looks fine at first glance.

Can you share a screenshot of the SK data path and value from the SK Data Browser and also from the KIP widget output and Widget Options please?

I'll take a look.

ESchlotfeldt, 2024-06-10:
Hi - thanks for quick response. below are the files requested. The issue might be that SK uses hPa as its default pressure unit, while KIP uses Pa as the default, ie KIP thinks it receiving Pa data and then converts it into hPa - which is out by a factor of 100 as observed in the screenshots?

[image]
[image]
[image]

godind, 2024-06-10:
I don't see the source emitting this value in your screenshot. The path is different than what's in the specs.
[image]

It's not a problem, but SK's base units type for all pressure values should be Pa, unless I am mistaken. You can ask in the specification channel to validate. 

KIP uses base Units as the starting point for all format conversions.

Until the source is corrected, in KIP, you can set the format to as-is value. This will pass it without modification, but the gauge displayed format will be As-Is. Not best looking but it will work.

ESchlotfeldt, 2024-06-11:
Ok thanks - thats very helpful!

godind, 2024-06-11:
Is it your own plugin that is generating the data?

Can I close this issue?

ESchlotfeldt, 2024-06-11:
Ok to close the issue. I hadn’t realised that Kip needs its input to be in Pa. I’ve changed to that and all is ok now. Thanks for your prompt assistance.

godind, 2024-06-12:
Always a pleasure to help. I'll close the issue.

Note: The base Units format is defined by Signal K. Not by KIP. See this lick for path specifications: https://signalk.org/specification/1.7.0/doc/vesselsBranch.html

closed 2024-06-12 by godind