mxtommy/Kip #430: Use SK perfomance data for close hauled (and downwind) angles

tkurki, 2024-06-15:
See discussion at https://github.com/SignalK/freeboard-sk/issues/149#issuecomment-2169158671

Kip should use (optionally?) the upwind targets from Signal K performance data instead of Kip's internal setting for close hauled angle.

If the system is deriving the target based on current wind conditions and polars the target will be dynamic, eg. updated just like any other value.

Does Kip show downwind targets in the wind widget? The same would apply there.

tkurki, 2024-06-15:
https://signalk.org/specification/1.7.0/doc/vesselsBranch.html?highlight=performance#vesselsregexpperformancebeatangle
https://signalk.org/specification/1.7.0/doc/vesselsBranch.html?highlight=performance#vesselsregexpperformancegybeangle
https://signalk.org/specification/1.7.0/doc/vesselsBranch.html?highlight=performance#vesselsregexpperformancetargetangle

godind, 2024-06-16:
Got it. Thanks.

godind, 2025-04-27:
I'll add SK plugin dependency feature first then get to this as I want to do another pass on the wind widget for improvements