mxtommy/Kip #431: Polar Performance plugin not updating in KIP

Techstyleuk, 2024-06-16:
Based on discussions in the Kip channel in Discord,  it seems that there is an issue with the data from the Polar Performance Plugin updating data in KIP.

See this video: https://youtu.be/vcKJ6G9lQtw

godind, 2024-06-16:
@Techstyleuk In KIP, do you see data from path performance.velocityMadeGood ? It looks like it's getting updated. And do you see data from the paths that don't get updates but have values?

Techstyleuk, 2024-06-18:
I will check Wednesday, when I am back on the boat

godind, 2024-06-21:
May I ask that you check if the plugin sends null values as the first values. We found a KIP Null value issue if it receive a Null as the very first value.

This may kill two bird with one stone.

Thanks you.

Techstyleuk, 2024-06-21:
I was on the boat last night and none of the values in the KIP data browser under signalk-polar-performance-plugin were updating, and none of them were zero or null, it did seem like they got one value and kept it.  SignalK data browser was updating.

godind, 2024-06-22:
@Techstyleuk can you confirm you are using `signalk-polar-performance-plugin`? There is also `signalk-polars-kraivio` that is a bit older. So far, it looks like there is an issue with the plugin. Still investigating...

linjon, 2024-06-22:
Hi, I’m having the exact same issue as Techstyleuk described in his video. I’m using signalk-polar-performance-plugin. If I use the Signalk built-in dashboard, the polar performance data is updating. But in KIP the data freezes the value it received when I start KIP. Thank you so mych for looking into this!

Techstyleuk, 2024-06-22:
> @Techstyleuk can you confirm you are using `signalk-polar-performance-plugin`? There is also `signalk-polars-kraivio` that is a bit older. So far, it looks like there is an issue with the plugin. Still investigating...

Yes, I am using `signalk-polar-performance-plugin'

godind, 2024-06-22:
@Techstyleuk Issue was fixed in the plugin by Hans. Update the plugin to v0.46 and it should work. May I ask that you report back and close the issue if fixed please?

Enjoy!

linjon, 2024-06-23:
Hi, I went to boat, updated and ran the Polar Performance plugin in KIP - it now works! Thank you so much! I’m not 100% familiar with issuehandling in Github, I suppose Techstyleuk should close the issue?

godind, 2024-06-23:
Thank you @linjon. Generally speaking, either the developer or que original user would close the issue. In this case I'd say it's safe to close without @Techstyleuk  feedback as the issue was not with KIP and both you and I have tested.

Thanks for the help and enjoy sailing!

closed 2024-06-23 by godind