mxtommy/Kip #437: Zones missing in KIP

dreuge, 2024-07-09:
I just installed the lasted Open Plotter 4 which is using Signal K  2.8.3 and KIP 2.12.0.   KIP no longer has "Zones", that is it is missing from the KIP configuration-> settings page top menu. Available menu items are: Signal K; General, Units, Datasets, & Storage.   Previous item "Zones" is not available.     Is this a know bug or user error?  Thanks

dreuge, 2024-07-09:
Issued resolved.  I was just informed of the new Zone setting in Signal K -> Data Browser: Meta Data [on].  By default Meta Data was off

godind, 2024-07-09:
Hi. Correct. Zones configuration has been retired from KIP in favour of SK defined zones.

Signal K zones are applied in KIP gauges and integrated in Notifications as before, with some minor improvements due to better integration.

I whish you a nice day!