mxtommy/Kip #440: Feature request: LED dimming slide

MrTom1, 2024-07-17:
Hi,
would be great to add the feature of a slider for LED dimming.

Regards,
Thomas

godind, 2024-07-17:
Hi. You mean a progressive value slider, right? Can you point us to a SK path that supports the type of value. I'm curious to know the what the specs are for this kind of data.

MrTom1, 2024-07-18:
[image]

godind, 2024-07-19:
Path info
- DimmingLevel path is ratio of 0  to 1 and supports PUT
- DimmingLevel of 0 or 1 will not emit state update of True or False
- check if state supports PUT
- Power published is watts consumption and read only

More spec detail to come...

godind, 2025-05-01:
@MrTom1 let me know when you are available to test this for me. I can build it but I can't test...

x7y8z9, 2025-08-02:
It would be my pleasure to support you with testing the integration of Shelly's into KIP. I currently have 68 Shelly's installed on board (Gen3 & Gen4). I would just like to point out that SK is running on an Ekrano as Victron OS Large, currently SK 2.13.5. My aim at the moment is to remain with the Victron standard in this regard.

I'm in switching only not dimming, incl. grouping of Shelly switches . 
Preferably KIP uses http commands like

Get status:
http://192.168.xxx.xxx/rpc/Shelly.GetStatus

Relay on:
http://192.168.xxx.xxx/relay/0?turn=on

Relay off:
http://192.168.xxx.xxx/relay/0?turn=off

Toggle:
http://192.168.xxx.xxx/relay/0?turn=toggle

as detailed by Shelly.guide for Shelly Relais/switches
https://shelly.guide/webhooks-https-requests/

godind, 2025-08-03:
Because KIP is primarily focussed on Signal K, I plan to use the Signal K paths rather than accessing direct API. This requires the use of the Shelly plugin (already available) but it also opens up lots of possibilities like automation using nodeRed.