mxtommy/Kip #453: signalk-avg-paths not working

Fgonzalez22, 2024-08-12:
I find very useful the signalk-avg-paths plugin. I tried to incorporate some of the paths provided by this plugin to KIP, but for some reason KIP only shows the first value it gets from signalk and stops updating.

godind, 2024-08-12:
I don't know this plugin. Can you check in both SK and KIP Data browser that you do get value updates for those paths?

godind, 2024-08-12:
Ôté that you can also use KIP's sample rate per individual path. This will smooth out data flow

Fgonzalez22, 2024-08-13:
> I don't know this plugin. Can you check in both SK and KIP Data browser that you do get value updates for those paths?

Yes I can see values change in SK data browser and I can see the paths listed in KIP. In Kip I don't see values, but this happens with other paths and they work fine. Attached pictures.

 [image]
[image]
[image]

Fgonzalez22, 2024-08-13:
> Note that you can also use KIP's widget Sample Time per individual path. This will smooth out data flow 

That's great! The sample time is the time in miliseconds Kip refreshes data and when it does it shows the average of all deltas within that time frame?

godind, 2024-08-13:
That's not right. KIP's Data Browser should list values and units. Do you see an error in the browser console log?

The Sample Time is the delay at which you want path data to refresh - no average, just a  simple reading. If you just need to smooth out KIP widget refresh it works well up to 1.5, 2'ish  seconds. To do longer average you can use Datasets and Data Chart widget.

I find that to smooth out gauge visualization and jitter, average calculation is not meaningful or impactful over that short periods. If you need over 2 seconds then it becomes somewhat meaningless.

Wind angles will moves fast but only a few degrees and so reading every 1 - 1.5 sec works very well, unless you're doing meteorological surveys ;)

If the data moves so much and the fast that it requires an average to be meaningful, something is wrong.

Fgonzalez22, 2024-08-13:
> That's not right. KIP's Data Browser should list values and units. Do you see an error in the browser console log?
>
> The Sample Time is the delay at which you want path data to refresh - no average, [...]

I can confirm that in kips data browser most paths don't show units or values. But still kip in its pages do show both. Seems to be an issue with displaying them in the data browser window. I don't have programing knowledge but managed to get the following from browser console:

[Authentication Service] User session token found in Local Storage
main.e0fab1a5a521589b.js:1 [Authentication Service] Deleting user session token
main.e0fab1a5a521589b.js:1 [Connection Service] Connecting to: http://localhost:3000
main.e0fab1a5a521589b.js:1 [Storage Service] Service startup. AppData API set to: http://localhost:3000/signalk/v1/applicationData/
main.e0fab1a5a521589b.js:1 [Delta Service] New endpoint: WebSocket opening...
main.e0fab1a5a521589b.js:1 [Delta Service] WebSocket connected without Authorization Token
main.e0fab1a5a521589b.js:1 [Connection Service] Server Name: signalk-server, Version: 2.8.3, Roles: ["master","main"]
main.e0fab1a5a521589b.js:1 [Authentication Service] User pantera login successful
main.e0fab1a5a521589b.js:1 [Authentication Service] Session Authorization Token received. Token Expiration: Wed Aug 14 2024 09:41:22 GMT-0300 (Argentina Standard Time)
main.e0fab1a5a521589b.js:1 [Delta Service] New token: WebSocket closing...
main.e0fab1a5a521589b.js:1 [Delta Service] WebSocket closed
main.e0fab1a5a521589b.js:1 [Storage Service] Retrieved config [default] from [user] scope
main.e0fab1a5a521589b.js:1 [AppInit Network Service] Initialization completed
main.e0fab1a5a521589b.js:1 [AppSettings Service] Service startup...
main.e0fab1a5a521589b.js:1 [AppSettings Service] Remote configuration storage enabled
main.e0fab1a5a521589b.js:1 [Dataset Service] Auto Starting 5 Datasets
main.e0fab1a5a521589b.js:1 [Dataset Service] Starting Dataset recording process: 1ddef575-1e39-4f47-bb6b-577ff32408d7
main.e0fab1a5a521589b.js:1 [Dataset Service] Path: self.electrical.solar.288.panelPower, Scale: minute, Period: 30, Datapoints: 1800
main.e0fab1a5a521589b.js:1 [Dataset Service] Starting Dataset recording process: f793f022-3141-41a0-a1f9-50be5f9f2b36
main.e0fab1a5a521589b.js:1 [Dataset Service] Path: self.environment.wind.directionTrue, Scale: minute, Period: 10, Datapoints: 600
main.e0fab1a5a521589b.js:1 [Dataset Service] Starting Dataset recording process: c746119a-952d-438a-a86b-ee24b5080d63
main.e0fab1a5a521589b.js:1 [Dataset Service] Path: self.environment.wind.speedOverGround, Scale: minute, Period: 10, Datapoints: 600
main.e0fab1a5a521589b.js:1 [Dataset Service] Starting Dataset recording process: f21ecad7-9c57-4229-ae66-806ba18b900a
main.e0fab1a5a521589b.js:1 [Dataset Service] Path: self.environment.wind.directionTrue_smooth, Scale: minute, Period: 10, Datapoints: 600
main.e0fab1a5a521589b.js:1 [Dataset Service] Starting Dataset recording process: 58a833a5-d148-4015-9676-5622bf1b667e
main.e0fab1a5a521589b.js:1 [Dataset Service] Path: self.environment.wind.speedTrue_smooth, Scale: minute, Period: 10, Datapoints: 600
main.e0fab1a5a521589b.js:1 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
Ui @ main.e0fab1a5a521589b.js:1
main.e0fab1a5a521589b.js:1 [Delta Service] New token: WebSocket opening...
main.e0fab1a5a521589b.js:1 [Delta Service] WebSocket connected with Authorization Token
main.e0fab1a5a521589b.js:1 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
_autoResume @ main.e0fab1a5a521589b.js:1
main.e0fab1a5a521589b.js:1 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
_autoResume @ main.e0fab1a5a521589b.js:1
main.e0fab1a5a521589b.js:1 [Connection Service] Server Name: signalk-server, Version: 2.8.3, Roles: ["master","main"]

godind, 2024-08-13:
It does not show any problems. I'll check the code with the plugin and data browser issue.

godind, 2024-08-14:
I fixed the Data browser bug.

I still searching the avg plugin data issue.

godind, 2024-08-14:
I also noticed you are using dataset with AVG plugin data. This will create averages of averages. You should use one or the other.

I don't think that for KIP widgets AVG plugin is very useful.

1- You can use Sample Time to visually smooth out data

2- Use Datasets for a combination of realtime, moving average and dataset average. If you use AVG paths here it will mess up the math.

Fgonzalez22, 2024-08-14:
> I also noticed you are using dataset with AVG plugin data. This will create averages of averages. You should use one or the other.
>
> I don't think that for KIP widgets AVG plugin is very useful.
>
> 1- [...]

Yea, you're right about avg o avg if using datasets. Just wanted to see how it would show up, jaja, because wind direction and speed was very chaotic. 
But I have been using datasets the way you mention and its very useful !

What about the dataset vertical graph....Its no longer in the latest version? And I cannot find the option of using simple linear with no progress bar. Its gone or can't I find the option?
 
Back to avg  plugin, I was planning as well in using some smooth avg paths as a source for derived data custom calculations. Just an idea for now. And I like the beaufort scale path and the wind quadrant info path( ie SSW).

godind, 2024-08-14:
Data Chart vertical graph has not been built. Not sure it brings a lot of value. We will need vertical for specific sailing performance widgets like average heading/AWA and VMG with polar. But that should it's own widget even if using the same Chart library.

I will check why the Simple Linear  has lots its No progress bar option. It should be there.

For AVG plugin, I looked at the plugin code, simple really. It sends both meta and values every time. Meta is only needed once. I think this might be the problem as KIP does not get the value updates past the first one.

From memory, SK should either send a Meta or a Value delta update but not both in the same delta. I'll dig this further.

godind, 2024-08-14:
The option to show No Progress is with the Linear gauge and not available in Simple Linear.

Fgonzalez22, 2024-08-14:
> The option to show No Progress is with the Linear gauge and not available in Simple Linear, but there is a bug. Fixed also...

Great! thanks for that as well!

PR #454 (2024-08-15): Godind/issue453
* Linear widget "No Progress" color option missing.
* Data Browser source and value not displayed. 
* Processing of combined values and meta delta updates. Fixes #453