mxtommy/Kip #455: Very difficult to control split line

Mastiffen, 2024-08-18:
When I have work on setting up a split screen, the split line follows the mouse even when the mouse button isn't pressed. Other times I can't even drag the split bar while holding the moues button down. Is this a known thing? I had to fiddle around until the split was right and then lock the layout by carefully moving the mouse out of the screen, away from the split line and TAB to the hamburger menu and chose "lock layout". It's almost like the split line thinks that I am pulling it down or up with my finger on a touchscreen.

godind, 2024-08-18:
Hi. It's the first time we heard of this problem. What hardware, OS and software are you running?

Mastiffen, 2024-08-18:
Impressively fast answer, thanks! MS Surface, Logitech MX705 mouse, Dual Dell U2717D on Displayport in over/under configuration, Windows 10, Chrome latest version. SignalK is running on a Pi 4 in my boat, Docker, latest version.

Mastiffen, 2024-08-18:
https://github.com/user-attachments/assets/20cb12e6-e5ab-4a9e-b2cf-887dfc0e8e7c

Mastiffen, 2024-08-18:
This is what I meant at the end of the last post:

https://github.com/user-attachments/assets/7096220c-e56a-48ec-9e12-0a5aaa12e7bf

Mastiffen, 2024-08-18:
I am on another laptop now, and I do not have the same problem there. Can it be something that happens wit a dual screen setup?

godind, 2024-08-19:
Possible. Or it's something to do with how windows/Surface handle touch and mouse capture.

godind, 2024-08-22:
Hi. Any update on this issue?

Mastiffen, 2024-08-22:
Not really, I stopped using it on the Surface with that setup. I just now tested it on the same surface, with the same everything except for monitor (this time with a 32" monitor in portrait configuration), and that doesn't have the problem. So I guess it's something with the resolution on the Dells, or something like that. Anyway, closing, because this is probably one of those freak things that nobody else will ever stumble over.

closed 2024-08-22 by Mastiffen