mxtommy/Kip #456: Load Predefined configuration button missing

bwssytems, 2024-08-21:
I am trying to load a saved configuration into another browser. In version 2.12.3, I do not see the button for "Load predefined configuration or rest configurations to defaults." This is using Brave Browser with the shields off and Microsoft Edge.

godind, 2024-08-21:
Hi. Thanks for reaching out.

Where did you save the configuration and with what version?

I suggest you look at the Help on Configuration Management. This will give you background on handling KIP config.

Let me know if you need more support. You can also join the KIP Discord channel where other users can assist.

bwssytems, 2024-08-22:
I am using SignalK 2.8.3 and KIP 2.12.3. I authenticate KIP with a token in signalk and it is in the Admin scope. When saving the config, I choose the Global area and give it a name. I cannot see or retrieve this in another browser from a separate machine or even the same machine.

godind, 2024-08-23:
Both users need to be admin to r/w to global. You can also use the same user and it will load the configuration automatically

closed 2024-08-24 by godind