mxtommy/Kip #474: Feature request: support keyboard to cycle trough page layouts

davidsanner, 2024-09-19:
Swiping R/L works great on tablets but having the ability to cycle through page layouts via std keyboard (arrow keys, space bar and/or pressing a corresponding number key) would be a nice feature.

EdKok, 2025-01-09:
Came to make this feature request myself.
But would use it myself in a bit more complex way, namely with a IR-remote. My display is in the wheelhouse above the windscreen and out of reach.  Hence the need for a remote.
I have the IR-part working in the OS but sadly KIP doesn't support keys (yet :) )

godind, 2025-03-09:
Keyboard shortcuts are planned for v3 as the new user experience is touch-first and gesture with fullscreen dashboards getting rid of all buttons.

closed 2025-04-15 by godind