mxtommy/Kip #482: Simple text new Widget request

hugorobi, 2024-10-28:
Hi,

Can you add a new widget that will allow us to just add specific text in a windows (not based on a SK value) ?

This will allow us to have a column or row title.

Thank !

[image]

godind, 2024-10-29:
Hi,

You can use the Basic Widget for this. When you a new split area, it has the Basic widget. Edit its properties and you can add a title.

hugorobi, 2024-10-30:
Hi,

I tried your suggestion. It worked, but unfortunatly, the font size is a little bit small.

If there is a suggestion list, can you add my request to this list ?

godind, 2024-10-30:
This is the list. I tagged it as a feature request.

Thanks for your input!

Rjn9533, 2024-11-12:
please give us the ability to choose font, color and size for out text.   Also the ability to add borders the a group enties to separate different groups of gauges.

It would be nice to be able to set ranges on gauges and specify colors for each range without having to use grafana.

Alarm points and ability to sound alarms......

godind, 2024-11-12:
@Rjn9533 thanks for the suggestion. Grouping gauge will not be possible with the existing framework.

For gauge zones with colors and alarms, this is already possible with the use of Zones. Have a look at SignalK meta Zones. They are configured for each path.

godind, 2025-03-29:
Static Label widget comming in v3.

closed 2025-04-15 by godind