mxtommy/Kip #487: New option for radial and linear gauge

hugorobi, 2024-11-18:
Hi,

For the Radial and Linear base gauge, can you add an option to have only the needle, removing the "progress" yellow part of the gauge.

This would allow us to use them for rudder angle, battery current variations (consuming or charging amps), etc...

[image]
[image]

hugorobi, 2024-11-18:
I found the option in Color for the "Lineare" gauge. Can you move this option as a specific one (checkbox). We will be able to choose the color AND having or not the "progress" part of the gauge.

godind, 2025-03-09:
Salut Hugo,

Tu peux clarifier ta demande SVP? L'indicateur provient d'une librairie alors ce n'est pas super flexible.

Merci

godind, 2025-04-27:
C'est partiellement fait. Le widget Linear possède une option aiguille seulement dans la v3 de KIP. 

Pour le radial. Qu'est-ce qui cause problème avec la progress bar ? Tu peux faire un cadran RPM, non?