mxtommy/Kip #488: NMEA 2000 PGN 127502 support

ODavis62, 2024-12-04:
Hello, It would be nice if KIP could support standard NMEA 2000 digital switching, PGN 127502 switch bank control and 127501 switch bank status. I believe 127501 is already supported but not in association with 127502. The current PUT request isn't supported by much anymore.
Thanks,
Orlando

godind, 2024-12-05:
Hi
Thanks for reaching out. KIP only reads SignalK path and values. It does not know about PGNs, sentences and such.

You need to have the right SignalK app that does the conversion from/to a protocol, such as n2k PGN, to path values.

KIP will send 1/0 to the path for on/off switching.

Hope that helps!

ODavis62, 2024-12-06:
Hi David,

Thankyou for the response. Yeah I did some more digging and found that out.
There is another plug in that supports the 1/0 thing from a path. Got it
working great now.

Again thank you, and I absolutely love KIP. It is just fantastic.
Kind regards,
Orlando

closed 2025-03-09 by godind