mxtommy/Kip #489: Show editor/Save edits - not saving

Haimana, 2024-12-23:
Hello,

I'm trying to use the configuration editor by checking the "Show Editor" checkmark.
I changed the source of self.navigation.courseOverGroundTrue / WidgetWindComponent from default to ais.GP but "Save Edits" button does not save the modification.
I loaded and edited the demo configuration. I cannot directly edit the WidgetWindComponent because I don't have all the required paths available for the moment and the Save button is not available

Is this a bug or I'm doing something wrong?

Thanks,
Seb

godind, 2025-01-07:
Sadly the config editor only supports local storage configuration. Try it in local storage - not signed in.

Once good copy local storage to global, then to your user.

An easier way is to edit the json config file directly on the server under ~/.signalk/applicationData/kip/users/...

Haimana, 2025-01-07:
First and last? 😃😃😃
Anyway, thanks for reply, really appreciate it. 
Best regards

godind, 2025-02-20:
Yep. Probably the last user ;)

It was initially created to ease development and config evolution but not used. It's dead code.

closed 2025-02-20 by godind