mxtommy/Kip #491: feature requests after a season sailib with KIP

motamman, 2025-01-07:
These aren't issues so much as requests/ideas after using KIP for one season on the water. I am now working on next season's iteration of my boat brain and a fistful of new sensors.

In no particular order or importance:

1. Import and export buttons for quickly moving design JSON files back and forth between different signal k servers.
2. More color options in general. A custom color palette?
3. A font size override? 
4. Center, left, and right; top, bottom, and middle alignment?
5. Remove unitless as a label and make it blank or a custom label.
6.  The option for all text and numbers to reflect the corresponding notification zone.
7.  The option for the radials and linear gauge bars to reflect the corresponding notification zone. 
8. An LED metaphor that can take bool values and turn them on and on (with customizable colors, of course.) 

This is a super presentation layer. 

Many thanks,
Mo

EdKok, 2025-01-09:
1. Is possible via "Configuration" -> "Settings" -> Tab "Storage".
I'm not completely happy with it, but does work. Not happy because a small mistake can destroy the json file. That is why I make manual backups these days before fiddling with KIP.
2, 3, 4 I agree
8. Is possible with "Boolean Control Panel" as the widget and "Type = Indicator".
Nice, but lacks alignment (as in 4.)

May I add:
9. Possibility to Cancel out completely of configuration without writing to json.

motamman, 2025-01-09:
Thanks, EdKok. I completely missed the indicator. It is nice but for the aforementioned styling limitations.

I know how to move files back and forth from local to server but not from server to server. I can manually copy files, but it would be nice to click a button and save them to another location (and the inverse at the other end to load a file).

godind, 2025-04-17:
Good point. I'll do a pass on units and add mm, inch and no unit

godind, 2025-04-18:
Not sure I see value for mm and inch. Please reopen with a use case if ever.
Features part of v3.

closed 2025-04-18 by godind

motamman, 2025-04-18:
precipitation, for one