mxtommy/Kip #553: Weird sizing of text in widgets

GraffJosh, 2025-04-02:
The text in widgets doesn't fill empty space very well (I'll try to attach a picture).

Take for example: numeric widget that is wide and not tall, the text is very small (too small to read on some configurations). I think the text won't fill down when the widget is "landscape".

GraffJosh, 2025-04-02:
[image]

godind, 2025-04-03:
The only thing I can see is fixing its size but it won't scale and might not look good on all form factors.

GraffJosh, 2025-04-05:
One way (and I do not know how to do this) could be to turn the text vertical when the spacing is more appropriate.

godind, 2025-04-05:
But if the widget is short but wide, then vertical label text would have little space up and down so probably not much more readable. Plus reading vertically is not obvious for most.

I think in the end, once you know your dashboard board, the titles become irrelevant to some degree. The eye goes naturally to the value.

GraffJosh, 2025-04-05:
I agree with that well enough. I'd feel comfortable closing as not-going-to-fix

closed 2025-04-15 by godind