mxtommy/Kip #556: Feature Request: Toggle Switches should create the path they publish on

GraffJosh, 2025-04-03:
I am having the DARNDEST time getting toggle switches up and running when I really just want them to publish to mqtt. 

I don't think it makes sense to require expertise in node-red to get this feature to work.

okFduCi8nl6bRz, 2025-04-03:
Our system attempted to address this issue:

Added MQTT support to SignalK requests service and boolean switch component to allow direct MQTT publishing without requiring Node-RED. This includes:

1. Added MQTT configuration options to widget config interface
2. Added MQTT support to SignalK requests service
3. Modified boolean switch component to handle MQTT publishing

Users can now specify MQTT topics directly in the widget configuration.

You can examine the suggested modifications here: https://github.com/okFduCi8nl6bRz/mxtommy-Kip-556/commit/edf8dd3fe5decc00f98fbd31aa94bdef2e21e665.

> [!CAUTION]
> Advisory: This solution was generated using automated tools, and we strongly recommend reviewing the code thoroughly before implementation. The solution may be partial - please consider it as inspiration for your own implementation.

---

http://missionfixinternet.com

GraffJosh, 2025-04-03:
Wait what did chat GPT just make a feature PR for the request I made??

godind, 2025-04-15:
Wrongly tagged a code fix to this issue. Should have been #568.

PR #579 (2025-04-15): Fix Toggle Switches Boolean Control Panel - Push mode not not changing color with touchscreen
- Fixes #568
- Does NOT FIX #556: Bad tagging.
closed 2025-04-15 by godind