mxtommy/Kip #605: lost control of embedded webpages after release 3.1.0

themagnetica, 2025-04-29:
After upgrading to the last release 3.1.0 the embedded webpages in a Kip layout are no more usable. Either in desktop and in tablet environment. For exemple, my setup includes a frame with the Avnav navigation ECDIS that becomes insensible to the finger touches in tablet mode and to mouse in desktop mode. I suppose this is caused by the new touch interface of Kips. This generate an odd tough.

themagnetica, 2025-04-29:
My bad. i have just seen that navigation events must be allowed in settings. that introduces a new odd. The sweep interface become unusable. Any workaround?

godind, 2025-04-29:
Have you tried swiping, fast and long enough, or using keyboard hotkeys?

themagnetica, 2025-04-30:
> Have you tried swiping, fast and long enough, or using keyboard hotkeys?

In tablet mode there is no way to  get gestures having effect at all. Maybe because the embedded webpage uses the whole screen space. I could try to add a small window in the lower corner of the tablet's screen but since i can't use the gesture command is not possible to access the setup menu anymore.
About keyboard hotkey, is there a page where to find the list? I was not able to figure out which they are. that could be useful for the desktop device.

godind, 2025-04-30:
See KIP's help.

Share your KIP url and your embed url please.

themagnetica, 2025-04-30:
config file:
`{
  "app": {
    "configVersion": 11,
    "autoNightMode": false,
    "redNightMode": false,
    "nightModeBrightness": 0.27,
    "dataSets": [],
    "unitDefaults": {
      "Unitless": "unitless",
      "Speed": "knots",
      "Flow": "l/h",
      "Temperature": "celsius",
      "Length": "m",
      "Volume": "liter",
      "Current": "A",
      "Potential": "V",
      "Charge": "C",
      "Power": "W",
      "Energy": "J",
      "Pressure": "mmHg",
      "Density": "kg/m3",
      "Time": "Hours",
      "Angular Velocity": "deg/min",
      "Angle": "deg",
      "Frequency": "Hz",
      "Ratio": "ratio",
      "Resistance": "ohm"
    },
    "notificationConfig": {
      "disableNotifications": false,
      "menuGrouping": true,
      "security": {
        "disableSecurity": true
      },
      "devices": {
        "disableDevices": false,
        "showNormalState": false,
        "showNominalState": false
      },
      "sound": {
        "disableSound": false,
        "muteNormal": true,
        "muteNominal": true,
        "muteWarn": true,
        "muteAlert": false,
        "muteAlarm": false,
        "muteEmergency": false
      }
    }
  },
  "dashboards": [
    {
      "id": "isplitsx-xxxx-4xxx-yxxx-xxxxxxxxxxxx",
      "name": "Dashboard 1",
      "configuration": [
        {
          "id": "widgetno-1xxx-4xxx-yxxx-xxxxxxxxxxxx",
          "selector": "widget-wind-steer",
          "input": {
            "widgetProperties": {
              "type": "widget-wind-steer",
              "uuid": "widgetno-1xxx-4xxx-yxxx-xxxxxxxxxxxx",
              "config": {
                "filterSelfPaths": true,
                "paths": {
                  "headingPath": {
                    "description": "Heading",
                    "path": "self.navigation.headingTrue",
                    "source": "default",
[... 260 lines cut ...]
                  }
                },
                "showMax": false,
                "showMin": false,
                "numDecimal": 1,
                "numInt": 2,
                "color": "contrast",
                "enableTimeout": false,
                "dataTimeout": 5,
                "ignoreZones": false
              }
            }
          },
          "x": 2,
          "y": 7,
          "w": 2,
          "h": 2
        }
      ]
    }
  ],
  "theme": {
    "themeName": ""
  }
}`

godind, 2025-04-30:
Make it simple please.

themagnetica, 2025-04-30:
let me try, i have to figure out the best way to serve kip on line. Anyway when control is transferred to embedded webpages neither shortcuts works. what you think about the idea of to activate a small controlbar when the control is transferred from sweep to embedded webpage?

godind, 2025-05-01:
I have made a minor update today that helps with the Embed widget. You can now use relative URL in the embed. Something like `/myplugin/`. This will prevent issues when you load KIP on another device then the server. It will also allow you to load apps installed on your Signal K server -if they use the same port. Swipe and hotkeys work fine. If they don't for you, look at the browser console log. You will see errors about window or document not being defined, it's because you are loading embed from a different host URL.

You must use the same hostname and port number that KIP uses in the embed. If not, the Browser will block script injection and prevent KIP from adding swipe and hotkeys in the embed. This is the browser security and not KIP. To authorize any app to be loaded as an embed (inside an iframe) you need to ask each app developper to add a special HTTP header in their application that allow their app to load inside KIP. It is the app that needs to agree to be loaded inside KIP, not the other way around. 

Embedding is troublesome because of browser CORS security rules, to prevent hacking. The experience will be worst if the content of the Embed also uses touch and gestures.

I have no plans at the moment to create button navigation controls. Sorry.

I am closing this as not a bug.
Join the KIP Signal K channel if you need further assistance with your setup.

Thank you and enjoy KIP!

godind, 2025-05-01:
Oh and by the way, the best way to have KIP online is to use the Venus OS from Victron. It has Signal K installed and it has a portal that you can access from anywhere. Free and works like a charm... Join the Signal K Discord server to get help.

closed 2025-05-01 by godind

godind, 2025-05-01:
> is there a page where to find the list? I was not able to figure out which they are. that could be useful for the desktop device.

Yes. See KIP's Help

themagnetica, 2025-05-01:
> Oh and by the way, the best way to have KIP online is to use the Venus OS from Victron. It has Signal K installed and it has a portal that you can access from anywhere. Free and works like a charm... Join the Signal K Discord server to get help.

Thank you for the advice, i was giving a try to ngrok but I stepped in some limit, it doesn't allow to share subfolders.