mxtommy/Kip #606: Feature request : Component to switch Dashboard

macjl, 2025-04-29:
Hello,

With the new version of KIP, the method to switch between dashboards relies on touch gestures. This might be great for some devices, but not for others. For example, it doesn’t work very well on my RPi screen.

Could you add a new component to switch between dashboards? It could also solve the issue with embedded pages where touch gestures are not possible.

Optionally, you could also create other components to go full screen, open the settings, etc.

Sherpa423, 2025-05-12:
Hello

Forgive me if this has already been picked up/dealt with, I can't select specific dashboard pages from their URL. I have flows in Node-Red that select display pages based on conditions using the page URL. All these have stopped working, which is related to the new UI in version 3 update. Is this a feature request to re-instate this or have I missed something.

AAlbero, 2025-05-28:
Same situation here. In certain stations I have either a webpage with buttons that open a specific KIP page, or a node-red scheme that display pages based on conditions.
There is a feature request already listed.

godind, 2025-07-25:
@Rjn9533 its very easy with a mouse. Just click, drag in the direction and release.

Just like a touch screen where you touch and swipe. It takes 5 min to get use to it.

Sherpa423, 2025-07-26:
Do we have a way yet for programmatically switching screens? I use Node Red to bring up screens based on conditions, such as engine on, or under sail. I can't swap out the deck mounted waterproof screens for touch versions, not would I want to, just need the previous way of displaying via a page URL or equivalent. At the moment I'm having to stick with v2.13.5 until an alternative comes along. Any options yet?

godind, 2025-07-27:
@Sherpa423 that would be in #638

godind, 2025-10-15:
@Sherpa423 since v3.10, there is a built-in remote control from any KIP to any KIP. It's just one check box to enable.

macjl, 2025-10-15:
Hello @godind 
It works much better now. 
The only use case I still have where I can’t use touch gestures, is when I remotely control the Rpi screen through VNC.
But that may not be a common use case. So the issue can be closed.

Sherpa423, 2025-10-15:
> [@Sherpa423](https://github.com/Sherpa423) since v3.10, there is a built-in remote control from any KIP to any KIP. It's just one check box to enable. It's another option to change your deck mounted waterproof screens.

@godind, Thanks for the pointer, though I don't currently have need for remote control, it looks like a good feature. The fix on #638 resolved my issue and I'm now on KIP v4.01. Thanks for all the hard work you all put in. The latest iterations of KIP are outstanding.