mxtommy/Kip #615: Switch Panel Label dont work (Pathproblem again)

SignalKuser, 2025-05-01:
hello,

Signal K Server version 2.13.5
Kip version  3.1.4

Switch Panel Label (toggle/push/indicator) dont "see" Signal K paths, its empty.

Strange: After the update to 3.1.4  the (new created) Push Button shows two active old Node Red paths for a short moment, it was only one time.
There was an inactive node red path, i activated the path but not seen in switch panel label. Buttons path section show nothing. 

I tried 3.1.4 Buttons, Old Buttons from before 3x , 3.1.1. Buttons. A new Dashboard Site, an old Dashboard site.

I can see the Paths in Data Inspector. Path value is Boolean, but i tested numeric option too.

Greets/Christian

godind, 2025-05-01:
Did you read the help documentation?

Send me a data inspector screenshot of you path please.

SignalKuser, 2025-05-01:
[image]
[image]
[image]

godind, 2025-05-01:
Your self.test.kip_button is of type String. It needs to be Boolean.

The documentation covers detecting those problems.

The other two paths should be visible. Are they not?

SignalKuser, 2025-05-01:
yes. your right.
changed that,

msg.topic = "test.KIP_Button";
msg.payload = (msg.payload === 'false') ? true : false;

return msg;

atm it works. the other 2 path too.
Big thx so far.

great work.

godind, 2025-05-01:
Join Discord channel. Faster and you have other user users that can help sometimes.

Enjoy.

closed 2025-05-01 by godind