mxtommy/Kip #616: Feature Request: Support for Color Bands (highlights / ranges) in WidgetGaugeNgRadialComponent

Cfokke, 2025-05-01:
Description:

I’m developing a modular marine instrumentation system using Signal K and KIP. I’ve been working with the WidgetGaugeNgRadialComponent to display temperature data. While the gauge renders correctly, I’ve noticed that it doesn’t display color bands when I include highlights or ranges in the configuration.

Expected Behavior:

The gauge should display color zones corresponding to specified temperature ranges, enhancing visual interpretation. For example:

"highlights": [
  { "from": -5, "to": 0, "color": "#00BFFF", "label": "Freezing" },
  { "from": 0, "to": 10, "color": "#FFFF66", "label": "Chilly" },
  { "from": 10, "to": 25, "color": "#90EE90", "label": "Comfort" },
  { "from": 25, "to": 40, "color": "#FF7F7F", "label": "Hot" }
]
Current Behavior:

Despite including the above configuration, the gauge doesn’t display the color bands. There are no errors in the console, and the gauge functions otherwise as expected.

Request:

Please consider adding support for highlights or ranges in the WidgetGaugeNgRadialComponent to allow for color-coded zones, which would greatly enhance the usability of the gauge for monitoring critical parameters like temperature.

godind, 2025-05-01:
Hi,

I'm not too sure what your goal
is and how you are playing to achieve it.

Editing gauge values like you are is not supported. I'm guessing you are editing the configuration file. If that is the case, I recommend not doing so. You KIP to build the config.

Highlights are defined in Signal K using meta Zones. The gauges work fine and so display colour and such.

Cfokke, 2025-05-01:
Thanks very much for clarifying — that helps a lot. I was manually editing the layout JSON to try to define visual zones for a temperature gauge using ranges or highlights, which apparently isn’t supported at the widget level.

Based on your feedback, I now understand that color bands are rendered based on meta.zones in the Signal K path metadata. I’ll explore how to define those zones via Signal K, so KIP can display the highlights properly.

If there’s any example or best practice for setting meta.zones in a Signal K plugin or JSON file, I’d appreciate a pointer. Thanks again!

godind, 2025-05-01:
Hi glad I could help. You should not have to edit any json file. Unless if you get help from a developer with an issue.

You can use the obvious steps of read KIP's Help and Signal K's web site documentation.

For support, join the Discord server for Signal K. Lots of people with experience on different subjects.

GitHub is more bug reporting.

closed 2025-05-01 by godind