mxtommy/Kip #620: widget timing out

motamman, 2025-05-03:
[image]

In the upper left corner, there are three styles of gauges showing the same data from the same path. The values can range from 0 to 5.

On the right, you can see how the linear and measure gauge widgets are timing out, sending the value to zero (as shown) from 5. It stays at zero for a tick or two and then returns to 5.

None of the other widgets on the dashboard is misbehaving. 

I created a new dashboard and added the same three widgets without timeout issues.

Mo

godind, 2025-05-03:
What is the timeout value on those widget?

motamman, 2025-05-03:
You mean the ttl? It doesn't matter. Initially there was none. And I experimented with various values from 5 to 30 seconds. No difference.

godind, 2025-05-03:
That console message is only emitted by the widget TTL feature. That means it's reach a set limit somewhere.

There's at least one widgets that has a TTL. We can't know from the screenshot if it's just one or multiple widgets that are configured this way.

Keep in mind that the TTL is set at the widget level and not at the path. Meaning the setting not shared with other widgets.

The TTL timer starts when it is executed. So if you have 3 widgets 
With identical TTL, their timer will start one by one so you will have a slight difference between each - depends on the computer's speed.

If you set the TTL too close the to the average time SK takes to send refresh that paths data, which can vary from update to update, it depends on the root hardware sending data, you will have flickers.

I'm not sure what you are trying to achieve but I get a feeling it's for a different purpose than what the feature is mentioned for.

TTL is meant to set the paths value back to `null` value, if KIP does not receive data after a period of time.

Like if someone shuts down the GPS by accident, or if you turn off the engine and turn off the key, after a period your position will be null, preventing navigation errors, or the engine RPM will be zero/null and not stay at the last SK value of 500rpm.

motamman, 2025-05-04:
I was just looking to see which gauge I wanted to use, so I have them
side-by-side. This is a bug. Whether it is worth fixing is a separate
issue. The simple linear has a 5-second TTL. It never goes to zero because
the data is updated every second. The other two have no TTL, but an error
is thrown every five seconds, and they bounce to zero. All three share the
same label and path. I can recreate the error in other dashboards.

godind, 2025-05-04:
I get it now. I'll check what the problem is.

godind, 2025-07-08:
This problem is because in an effort to share ressources, path observer registration (paths used widgets) are unique and shared based on the path string. It's on a first come first serve basis so the first widget to register that path sets the TTL value for all others.

This explains why other widgets using the same path can't have different TTL setting.