mxtommy/Kip #621: Linear gauge, strange behavior for 0-260

macjl, 2025-05-03:
Hello,

I have a tank of 260l, and then created a linear gauge with range 0 to 260. But the gauge show 0 to 300. If I set 0 to 250, it’s working fine. Also working as expected for other gauge I have ( 0 to 40, and 0 to 160)

[image]
[image]

godind, 2025-05-03:
Thank you for this report. It's how the scales work. The gauge will generate nice scales base on the requested value. The scale have to be divided by a minimum amount of tick bars, each with values number that has no decimal so it's not always 100% what you asked for.

You can remove the Ticks on the Linear or use the  Radial in capacity mode. Also most path like tanks also have a percentage path.

macjl, 2025-05-04:
I removed the Ticks, but still same behavior : 

[image]

godind, 2025-05-04:
Hum... surprenant que personne n'est remarqué depuis 3 ans!

Je vais faire la correction! Ainsi que les deux petites bars grises. 

I'll fix the bug.

Thank you

macjl, 2025-05-04:
Merci !

PR #623 (2025-05-04): Linear gauge not respecting scales with no ticks
Fixes #621
commit 872ed070 (2025-05-04): Linear gauge not respecting scales with no ticks (#623)

Fixes #621
closed 2025-05-04 by godind