mxtommy/Kip #624: Text overlap on low resolution screens

macjl, 2025-05-04:
Hello,

On m’y Rpi screen, I notice some text overlaps. I think that’s because the screen has a quite low resolution (800x600).

You can see what I mean in the bottom line of the screen capture : 

[image]

macjl, 2025-05-04:
And I also have difficulties to show the Autopilot Component

[image]

godind, 2025-05-06:
What version do you have?

godind, 2025-05-06:
I've adjusted the canvas to account for scale size. That should bring Display name, min/max and units closer to the edges. It will still be a tight fit.

The AP Control Head is not responsive. Never been. It's something that needs to be redesigned. For now you have to use a large area.

godind, 2025-06-16:
@macjl could you beta test the new responsive AP widget for me? My boat is on the hard this season and so I cannot test.

godind, 2025-06-28:
Let finish a version first. I'll let you know when ready if I still need help. On a road trip at the moment so no coding ;)

Thanks again.

commit c110cc17 (2025-05-06): Text overlap on low resolution screens
Fixes #624