mxtommy/Kip #630: path for slider

searay24, 2025-05-11:
Could not select any paths Slider Control
could not see any data Slider Control

godind, 2025-05-12:
Hi,

Thanks for contacting us. First, go over the help documentation. The path needs to accept PUT commands (receiving data) for it to be visible.

Once the path supports, send a screenshot of the path in the Data Inspector and I'll try and find the bug.

Thank you

godind, 2025-05-12:
Also add your signal K server version information.

searay24, 2025-05-15:
thank you for your help my dash is fantastic the zones look great I have
Cat 3208 motors with parallel sensors and a cheep n2k converter
i'm just  messing with the n2k autopilot The problems im having is Auto
pilot state it doesn't show up in my menu only some signalk setting data
my signalk ver is 12.13.5 my auto pilot is a Yacht Devices YDAP-04 and a
home made Node Red compass  anda pwm  h bridge  im messing with that now

godind, 2025-05-16:
Hi,

Thanks for the positive feedback! I suggest you join the Signal K Discord Help channel for this kind of support. You will find more knowledgeable ressources there.

Good luck!

closed 2025-05-16 by godind

searay24, 2025-05-16:
Thank you i will one more quick question though please how do i get target
heading to display