mxtommy/Kip #631: Text is drawn/rendered twice

edvaldeysteinsson, 2025-05-12:
In at least v3.2.0(possibly earlier) all texts seem to be drawn/rendered twice

[image]
[image]
[image]
[image]

godind, 2025-05-12:
This is on iOS only, right?

godind, 2025-05-13:
It's a know Apple bug when using non system fonts. KIP uses Google's Roboto font.

The only fix is to change KIP to another a system font or wait for a fix.

edvaldeysteinsson, 2025-05-14:
This is in Safari on Mac

godind, 2025-05-14:
Yeah. Older KIP used Arial font which is not a problem.

godind, 2025-05-18:
Hum. I really like this font!!! Not sure how to fix it. It's only on Apple Safari (MacOS and iOS) so far. I've seem a few reports that mentions it's a long standing know issue. I've triple checked and the font is not drawn multiple time, nor is it resized in rapid succession or canvas not cleared before redraw. It also shows up is some SVG elements but not always...

JonyIvy, 2025-06-22:
Maybe an option in the config to use the old font until this is fixed? On my iPad and Mac the Arial looks better than the triple-Roboto...