mxtommy/Kip #636: Date/Time widget does not allow high or low tide time from signalk-tides plugin as data source

ryansabin, 2025-05-15:
The Date/Time widget does not allow you to input environment.tide.timeHigh or environment.tide.timeLow from the https://www.npmjs.com/package/signalk-tides plug in

godind, 2025-05-16:
Yeah. Without having looked deeply at this, the app owner needs to change the way the time value is handled.

Hopefully someone will take the issue back to him.

godind, 2025-05-19:
@ryansabin  Can you share a screen capture of the path and data from SK's Data Browser and KIP's Data Inspector?

ryansabin, 2025-05-20:
[image]

godind, 2025-05-21:
Can you also share the path from the KIP Data Inspector please.

I am 99% certain it's because Tides add a custom meta unit the dare string path. As per specification, strings data types do not require units, only numbers, and Tide's unit is not a Signal K unit so KIP thinks it's a number of unknown unit.

If the developer removes the meta unit on the date string, KIP will detect it as a date.

My guess is Tides it's self does not require the meta unit as it probably knows what to do with each path without a unit present.

Or it's something you added manually your self?

ryansabin, 2025-05-21:
[image]

ryansabin, 2025-05-21:
I have reported an issue with Tides developer https://github.com/bkeepers/signalk-tides/issues/6

Hopefully we can get this fixed:)

godind, 2025-05-22:
From your screenshot NaN means not a number. As expected.

tvr256, 2025-06-12:
Unfortunately I think this might be a KIP issue.

Looking at the SignalK spec, the Tide object uses the Timestamp definition for  timeLow and timeHigh: https://github.com/SignalK/specification/blob/master/schemas/groups/environment.json

[image]

The Timestamp definition uses units "RFC 3339 (UTC)", and this matches the SignalK data inspector:
https://github.com/SignalK/specification/blob/master/schemas/definitions.json

[image]

[image]

From what I can see, KIP doesn't recognise units of "RFC 3339 (UTC)" so it defaults to "unitless" instead:

[image]

godind, 2025-06-28:
KIP already has code to validate and convert UTC valid Stings to date object. This happens very early in the process and only for json string values.

Also important, the server uses the object schema definitions but most client app, like KIP, don't. 

Since KIP does not use the object schema definition, it relies on metadata information sent as delta updates by the server and dynamically build path meta/schema info.

KIP will look at the data type and if it is of type numeric, and only numeric, it will look at the meta unit value to determine conversion groups. If it receives no meta units for the path (meta not defined) it will be tagged as unitless and allows the user to pick from all conversions groups available.

The issue with KIP as it stands, and last I asked Scott, is no units officially exist for types other than numeric. So when KIP sees a unit for a path, it thinks the value is numerical and not a string. That's what is causing the issue.

As seen in the text below, copied from the specs, it's not clear if the string **RFC 3339 (UTC)** is a valid Units label. Note the word (undefined).

A UTC date time that works well in KIP with no meta units:
```
/vessels/<RegExp>/navigation/
datetime

Description: Time and Date from the GNSS Positioning System
```

One with issues that has meta units:
```
/vessels/<RegExp>/environment/tide/timeLow

Units: RFC 3339 (UTC) (undefined)

Description: RFC 3339 (UTC only without local offset) string representing date and time.
```

I'll ask in the Spec channel for confirmation.

If confirmed as valid, I need to add support for meta units over strings.

tvr256, 2025-06-28:
I'm not a schema expert, but it looks like most "units" are listed in the spec [here](https://github.com/SignalK/specification/blob/1978f556d8790be2b3bfdc9fffcf82a9a913e3c2/schemas/definitions.json#L87). But RFC 3339 (UTC) isn't included in the list, could this be why it's reported as (undefined) in the meta units?

godind, 2025-06-28:
Yes, it's not part on the units. But you can find it in the `timestamp` definition.

Where do you see it as undefined?

godind, 2025-06-29:
@tvr256 after looking deeper at this, if you look at the specs for timeLow and timeHigh, even if the reference the timestamp definition, they have no units property defined, only descriptions:
```
"timeLow": {
     "description": "Time of the next low tide in UTC",
     "$ref": "../definitions.json#/definitions/timestamp"
   },
"timeHigh": {
    "description": "Time of next high tide in UTC",
    "$ref": "../definitions.json#/definitions/timestamp"
  }
```

These that are part of the same do:
```
"heightHigh": {
          "description": "Next high tide height  relative to lowest astronomical tide (LAT/Chart Datum)",
          "$ref": "../definitions.json#/definitions/numberValue",
          "units": "m"
        },
        "heightNow": {
          "description": "The current tide height  relative to lowest astronomical tide (LAT/Chart Datum)",
          "$ref": "../definitions.json#/definitions/numberValue",
          "units": "m"
        },
        "heightLow": {
          "description": "The next low tide height relative to lowest astronomical tide (LAT/Chart Datum)",
          "$ref": "../definitions.json#/definitions/numberValue",
          "units": "m"
        },
```

That does suggest the should not have units, but that the numeric ones should.

godind, 2025-06-29:
@tvr256 but I still don'"t get where the metadata unit values is coming from... Have you looked at the tides plugin code?

If you remove the units meta value, it should work fine in KIP.

emonty, 2025-06-30:
There actually isn't any units in the tides plugin code. But if you follow the specification link for the timestamp: https://github.com/SignalK/specification/blob/master/schemas/definitions.json#L8-L15 you'll see that the units is attached to it there

```
    "timestamp": {
      "type": "string",
      "description": "RFC 3339 (UTC only without local offset) string representing date and time.",
      "units": "RFC 3339 (UTC)",
      "example": "2014-04-10T08:33:53Z",
      "format": "date-time",
      "pattern" : ".*Z$"
    },
```

The `numberValue` definition doesn't have `units` defined on it because there are multiple valid units a number could have.

In any case, the tides plugin isn't doing any of the definition of any of this - they're just sending the delta for `environment.tide.timeLow` and the units transitive through from the specification definition.

godind, 2025-06-30:
I've confirmed with Scoot that strings are not prohibited from having units. Tides is the first to do so...!

I'll need to add this feature.

godind, 2025-07-03:
@emonty data.service.ts is the right place.

- Things to know: skData is the local data tree with values and meta. The unique key to locate data is the path.
- pathItem.type represents the javaScript data type and not the signal K meta units. It helps KIP filter paths on data types rather then Units, which are not always defined....well very often not.

There are two ways paths can be added to skData depending on what type of data/info the server sent first:
1.  value received first - see method `private updatePathData()`
2. metadata received first - see method `private setMeta()`

The server always tries as best as possible, to prioritize metadata information first. Because the paths in question are part of the schemas, the server sends those meta before the values. So we are on options 2.

Look at this part in setMeta(): `type: meta.meta.units ? "number" : undefined`. It's where if meta has units it's always a number data type. We need to add "Date" to the type if unit is "RFC 3339 (UTC)", else "number". Other types don't have units... yet.

This is where if the value came in first, it parses all **new paths** that are String once, and determine if they are RFC compatible.
[[if (pathType === "string" && isRfc3339StringDate(dataPath.value)) {](https://github.com/mxtommy/Kip/blob/b564605ddd0830b0b876f735a3f2795e2997efaf/src/app/core/services/data.service.ts#L275-L277)](https://github.com/mxtommy/Kip/blob/b564605ddd0830b0b876f735a3f2795e2997efaf/src/app/core/services/data.service.ts#L275-L277)

We need to keep this because some paths do have RFC string but they are not in the schema. It's a KIP helper we could remove in the future when SK schema is cleaner.

This is where in `private updatePathData()` things go wrong. It's not undefined, it should be "Date".
[if (pathItem.type === undefined && dataPath.value !== null) {](https://github.com/mxtommy/Kip/blob/b564605ddd0830b0b876f735a3f2795e2997efaf/src/app/core/services/data.service.ts#L300-L305)

I think that's it. We don't need to convert the value as it's up to widgets to convert to Date(). Lots of testing should be done... it's THE core KIP service.

emonty, 2025-07-03:
First draft pushed, this works locally when deployed into my local signalk. I have not added any explicit tests yet - just wanted to push it up and make sure this general direction looks ok. Thanks for the pointers!