mxtommy/Kip #638: Feat: Add dashboard number to URL string

godind, 2025-05-16:
Make it possible to select a specific dashboard using the URL. This also adds he ability to bookmark dashboard views.

tkurki, 2025-07-26:
this would make KIP much more useful as a Freeboard sidebar, where you want a tall and narrow layout.

godind, 2025-07-27:
Hi @tkurki,

From what I know, it works well with Freeboard-SK. Tell me more about how you see this feature helping the integration?

closed 2025-07-27 by godind

godind, 2025-07-27:
@tkurki in the next release you can use dashboard/{id} as in: `/@mxtommy/kip/#/dashboard/3`, where dashboard is an array id from 0 to x dashboard.

Note that swiping over KIP, in all four directions, should also work.

tkurki, 2025-07-27:
Nice!

motamman, 2025-07-27:
This is really a wonderful addition.

PR #726 (2025-07-27): Add dashboard number to URL for bookmarking
Enable selection of specific dashboards via URL, allowing users to bookmark their dashboard views. This change enhances navigation by integrating dashboard IDs into the routing structure.

Fixes #638